tock/libtock-rs #2: [RFC] Consider creating a tock-quickstart template

japaric, 2017-05-10:
like the [cortex-m-quickstart] one.

[cortex-m-quickstart]: https://github.com/japaric/cortex-m-quickstart

Since building an application requires some information stored in files, like platform specific linker scripts and target specification files, that Cargo can't pass from a dependency to a top crate, I suggest using a Cargo project template to provide these files.

The workflow I'm picturing would look like this:

```
# grab the Cargo template
$ cargo clone tock-app-template --vers 0.1.0

# rename the directory
$ mv tock-app-template my-app && cd $_

# change the project information
$ edit Cargo.toml && head Cargo.toml

[package]
authors = ["Jorge Aparicio <jorge@japaric.io>"]
name = "my-app"
version = "0.1.0"

# configure the template (not sure if actually needed)
$ edit memory.x && cat memory.x
MEMORY {
    FLASH (rx) : ORIGIN = 0x80000000, LENGTH = PROG_LENGTH
    SRAM (RWX) : ORIGIN = 0x00000000, LENGTH = RAM_LENGTH
}

# depend on libtock
$ cargo add tock --vers 0.1.0

# write the app
$ edit src/main.rs && cat $_
#![no_std]

extern crate tock;

use tock::{led, timer};

fn main() {
    let led_count = led::count();
    loop {
        for i in 0..led_count {
            led::toggle(i as u32);
            timer::delay_ms(500);
        }
    }
}

$ xargo build --target thumbv7em-tock-eabi --release

$ tockloader install target/thumbv7em-tock-eabi/my_app
```

The template would contain, in its root, target specification files like `thumbv7em-tock-eabi.json` and a linker script (`memory.x`, in this example) for configuring some aspects of the memory layout of the program (I'm not sure if this required for Tock apps since they are relocatable).

The `tock` crate would have a build script (like [this one](https://github.com/japaric/cortex-m-quickstart/blob/7fa559eeab98bb21f7378c4344da81b805b823d8/build.rs)) to pass the main portion of the linker script (currently, `layout.ld` in this repo) *to the top crate* such that the linker can find it when linking the application.

alevy, 2017-05-10:
Yes, I am 100% for this! I wonder if it makes sense to flesh out the the library a bit more before putting effort into this, but definitely makes sense.

RE: The linker script, it is currently required because of how the `elf2tbf` tool (which sticks a Tock-specific header on the binary) expects to find sections in the ELF, but I believe some minor changes to that might obviate the need for a linker script.

torfmaster, 2018-03-23:
A quick start can be achieved by adding an example and use the run_example script. I thinḱ we should close this issue in favor of more concrete issues concerning the usability of libtock-rs. Any opinions?

Woyten, 2018-03-23:
I am also for closing this issue as this has been open for a very long time. Instead, we could add some instructions to the Readme on how to write a custom example to evaluate libtock?

PR #29 (2018-03-30): Added a quick-start section to the readme.
This PR creates a "quick-start" section to Readme.md. I created this PR to satisfy #2. Currently, the build infrastructure consists of the following elements
 - Cargo.toml
 - xargo.toml
 - the target json
 - rust-toolchain
 - tockloader
 - elf2tbf
 - layout.ld
 - a short shell-script

Any of these components is subject to constant change and perspectively I expect the following could happen
 - xargo will be replaced by built-in functionality of cargo
 - elf2tbf will land (as a standalone application) on crates.io
 - the shell-script will be replaced b [... cut]
closed 2018-03-30 by alevy