tock/libtock-rs #8: Can't get example app to work

torfmaster, 2017-11-29:
Hi, 

thank you for developing libtock-rs. I'm trying to get an example app (blink) to work on my nrf52-dk. However, I don't get any reaction and I have no idea where the mistake may be.
I tried the following:
 - Fix thumbv7em-eabi.json by adding:
`
	"target-c-int-width": "32",
`
(I'm using rustc 1.23.0-nightly (d0f8e2913 2017-11-16)).
 - Compile (release-mode) using xargo:

`
xargo build --examples --release --target thumbv7em-tock-eabi
`

 - Use elf2tbf to package the artifact:
`
elf2tbf --include-pic-info -v -n blink -o cortex-m4.bin blink
`

Here I get the following output:
`
          version:        2        0x2
           header_size:       88       0x58
            total_size:      512      0x200
                 flags:        1        0x1
        init_fn_offset:        1        0x1
        protected_size:        0        0x0
      minimum_ram_size:     2048      0x800
           text_offset:       88       0x58
           data_offset:      270      0x10E
             data_size:        0        0x0
     bss_memory_offset:        0        0x0
              bss_size:        0        0x0
relocation_data_offset:       88       0x58
  relocation_data_size:        0        0x0
            got_offset:      270      0x10E
              got_size:        0        0x0
  minimum_stack_length:        0        0x0
    flash region:
                offset:       88       0x58
                  size:        0        0x0
`

 - Make a .tab out of the tbf
 - Upload the .tab using tockloader:
`
tockloader install --jtag --arch cortex-m4 --board nrf52-dk --jtag-device nrf52 --app-address 0x20000
`

- Press "reset".

The same workflow (except for I am compiling C-apps) works for the userland/example/blink-app and is successful.

Installing the rust-app works fine on the board, tockloader finds it. But it doesn't run the app properly (i.e. doesn't blink, neither does hello).
The output of the board (tockloader listen) is:
`
TOCK_DEBUG(0): src/main.rs:343: Initialization complete. Entering main loop
`
Do you have any idea, where the problem is?

alevy, 2017-11-29:
@torfmaster The process you went through seems right. The version of this library is currently tailored to the Hail board, which puts applications in a different location in RAM (0x30000). This is a limitation of the library that needs work---short version it has to do with figuring out how to properly relocate the vtables that Rusts generates.

I believe the fix is to modify the linker script in your app to put the code at `0x20000`, but I will test on the NRF52dk when I can and confirm exactly what to do.

niklasad1, 2018-02-24:
Jepp, I get the same behavior as @torfmaster on ```nRF52-DK```

torfmaster, 2018-02-24:
@niklasad1 we have forked libtock-rs. We have some basic functionality working on nrf52-dk as led, buttons, gpio, temperature, simple_ble. The run_example script should be just fine for testing.

alevy, 2018-02-25:
@torfmaster the new relocation makes this harder to solve with small tweaks to this repo, but is going to help write a much more robust version. That's on my todo list, or at least to sketch what needs to happen for that.

torfmaster, 2018-02-26:
@alevy we adapted our fork https://github.com/torfmaster/libtock-rs to work with the new relocation. We don't have global variables, but for the moment we don't need them. I would be interested in your opinion.

torfmaster, 2018-03-12:
I think I can close this issue for now.

closed 2018-03-12 by torfmaster