tock/libtock-rs #24: New official collaborators

alevy, 2018-03-18:
@torfmaster and @Woyten, since you have been doing the lion's share of work bring this library up to speed, do you want to be added as contributors?

This would basically mean you'd be able to merge pull requests yourselves. I think the one thing we'd want us just to make sure there is a good way to contact you (e.g. that you are on the mailing list).

There is no formal process for this library (yet), but managing it core-team-esque, where mostly big design decisions, as well as things like when to do a release version, are done by consensus seems like a good idea.

Woyten, 2018-03-18:
Hi @alevy, thank you very much for appreciation! :slightly_smiling_face:

Having the ability to merge PRs ourselves would help a lot integrating new features easily! However, for non-trivial, possibly breaking changes I would still want to wait for one of the Tock pros to start and approve a code review. This is the process anyway, right?

Regarding the mailing list, you can just add me.

I am only speaking for myself here. Thus, before adding @torfmaster I would wait for his feedback.

alevy, 2018-03-18:
@Woyten you can add yourself to the mailing list (it's just a Google group): https://groups.google.com/forum/#!forum/tock-dev it's not a closed group or anything.

The process for the kernel/c-userland repo is for major changes to be approved by all the "core team". I

alevy, 2018-03-18:
@Woyten you can add yourself to the mailing list (it's just a Google group): https://groups.google.com/forum/#!forum/tock-dev it's not a closed group or anything.

The process for the kernel/c-userland repo is for major changes to be approved by all the "core team". We can do something similar for this repo, though I think playing it by ear for now seems fine. Basically, if you are not comfortable merging something, yeah, totally fine to wait to hear from one of us.

torfmaster, 2018-03-19:
Hi @alevy,

thank you for this proposal. I would like join as collaborator as well.

closed 2018-03-19 by alevy