tock/libtock-rs #27: Safe Concurrency

Woyten, 2018-03-23:
I am currently trying to make the callback API more thread safe by via `SubscribableCallback: Send`.

As expected, I am not able to share any mutable state between a callback and the main thread. This would require a type like `std::sync::Mutex` which is not available for Tock.

My idea would be to write some simple Tock compatible Mutex type that manages a lock and yields when a lock situation is encountered.

What do you think? Is this a good idea? Is it easily feasible? Maybe there is some library support for it?

brghena, 2018-03-23:
In Tock, the architecture of application callbacks is such that callbacks only occur after the application calls `yield`. They can never occur in the middle of data access. This means a mutex shouldn't actually be necessary for data items used in both the callback and main thread, since their execution is already mutually exclusive as guaranteed by the OS.

Is the problem here that we have no way of specifying to Rust that data accesses are thread safe, or am I misunderstanding the problem?

Woyten, 2018-03-23:
Thanks for your response! To be honest, this surprises me. Could you have a look at the `button_subscribe` example? We loop at the end of the program without yielding. The callback is executed whenever you press a button. Doesn't that mean that the main thread is interrupted?

Woyten, 2018-03-23:
Or is it because `sleep` is yielding? Hmm...

brghena, 2018-03-23:
Yup! `sleep` yields.

Woyten, 2018-03-23:
Thank you very much. So, we never have to deal with concurrency, right?

Woyten, 2018-03-23:
Closed because the Tock architecture is cool!

closed 2018-03-23 by Woyten

brghena, 2018-03-23:
I would be careful though. This architecture definitely does not solve all problems. Particularly, if you `yield` inside a callback, you can end up re-entrancy problems which may need some form of locking.

Woyten, 2018-03-23:
Is it a problem we can solve in Rust?