tock/libtock-rs #32: When should Rust or Tock be updated?

Woyten, 2018-03-30:
Currently, @torfmaster and me are tracking builds of libtock-rs against the most recent nightly toolchain on https://travis-ci.org/torfmaster/libtock-rs/branches. Once we see that the build starts failing, we fix it, update the nigthly toolchain and create a PR. In this way, we avoid that new code that is considered to be valid right now becomes invalid very early due to an update of the toolchain.

However, recently, the breaking builds emerged quite frequently and I fear that users of libtock-rs might feel discouraged when needing to update their toolchains again and again. The same applies to updates of the Tock submodule.

My proposal:
- We use the same toolchain as Tock (currently 2018-03-07) s.t. people only need a single toolchain
- We update Tock **and** Rust when the toolchain of Tock is updated
- We update Tock when an important feature (bugfix, enabler, ...) has been merged to Tock

alevy, 2018-03-30:
This repo is only relying on the Tock repo for `elf2tbf`, right? So that's probably the only important component to track changes for.

Nonetheless, I agree that keeping the rust version in-line with Tock's makes sense for exactly the reasons you listed.

Woyten, 2018-03-30:
`elf2tbf` is one reason. Another reason is that the kernel gets incompatible to libtock-rs from time to time. The user should not lose time searching for the correct kernel version but just use the kernel of the submodule.

Do you think, I should change the nightly back to 2018-03-07?

Woyten, 2018-04-13:
My last question is very old now. The next updates will arrive as soon as `xargo` less compilation works and/or `elf2tab` is merged.

closed 2018-04-13 by Woyten