tock/libtock-rs #74: Issues running optimized apps

torfmaster, 2019-02-08:
I'm facing issues running apps with opt-level=0,3. Even trivial examples like hello don't run and there is no feedback neither from the kernel nor from tockloader/elf2tab. I currently no idea at which point of the toolchain the problem occurs.

Possible candidates:
 - the binary itself is corrupt/too big
 - it cannot be flashed
 - elf2tab creates incorrect tabs

Any help is highly appreciated!

jrvanwhy, 2019-02-08:
My recommendation is to create a debugging system call similar to what we have at https://github.com/google/tock-on-titan/blob/master/golf2/src/debug_syscall.rs. You can then manually call the system call from the entry point's assembly as follows:

```
// Overwrite r0 with the driver number (save r0 manually if necessary)
movw r0, #1
movt r0, #0x8000
svc 2
```

torfmaster, 2019-02-08:
Great, thank you!

rajivr, 2019-02-08:
Previously when trying to do non-optimized builds for the kernel, I encountered stack overflow issues. Since then I always use `opt-level = z` as default. I am now doing the same for user-space `Cargo.toml`.

If I want to un-optimize a variable, I use the following trick.

```
asm!("" : : "r"(xxxx));
```

which forces the compiler not to optimize out the variable `xxxx`, allowing us to inspect it in the debugger.

jrvanwhy, 2020-06-09:
@torfmaster is this issue obsolete?

jrvanwhy, 2020-07-07:
I think this issue is obsolete.

closed 2020-07-07 by jrvanwhy