tock/libtock-rs #82: `minimum_ram_size` has an extra 2048 bytes

rajivr, 2019-02-24:
In `layout.ld` we add `STACK_SIZE` bytes to `.data` section

```
    .data :
    {
        /* Account for MPU alignment and the stack */
        . = ALIGN(MPU_MIN_ALIGN);
        . = . + STACK_SIZE;
        . = ALIGN(8); /* The stack is aligned to a multiple of 8 bytes. */
        _data = .;
        KEEP(*(.data*))
        . = ALIGN(4); /* Make sure we're word-aligned at the end of flash */
    } > SRAM
```

Based on my understanding of `userland_generic.ld` in `libtock-c`, it seems like `elf2tab` assumes that the stack is allocated outside of `.data` and `.bss` section.

For rust apps, this is currently leading to a situation where we are allocating an extra 2048 bytes. Using `blink.rs` as an example, we can see the following.

```
$ arm-none-eabi-size target/tab/blink/cortex-m4.elf
   text    data     bss     dec     hex filename
   1228    2048      20    3296     ce0 target/tab/blink/cortex-m4.elf

$ elf2tab -n blink -v -o target/tab/blink.tab target/tab/blink/cortex-m4.elf \
    --stack 2048 --app-heap 1024 --kernel-heap 1024 --protected-region-size=64
Min RAM size from sections in ELF: 2068 bytes
Number of writeable flash regions: 0
Including the .crt0_header section at offset 64 (0x40)
Entry point is in .text section
Including the .text section at offset 128 (0x80)
Including the .data section at offset 1276 (0x4fc)
Searching for .rel.X sections to add.
TBF Header:
               version:        2        0x2
           header_size:       44       0x2C
            total_size:     4096     0x1000
                 flags:        1        0x1

        init_fn_offset:       85       0x55
        protected_size:       20       0x14
      minimum_ram_size:     6164     0x1814
```  

From `arm-none-eabi-size`, we can see `.data` is 2048 bytes, which is basically memory for stack and `.bss` is 20 bytes.

`elf2tab` correctly identifies _Min RAM size from sections in ELF_ to be _2068_ bytes.

The _2068_ bytes includes _2048_ bytes of stack memory. Therefore _minimum_ram_size_ should be _2068 + 1024 (`--app-heap`) + 1024 (`--kernel-heap`) = 4116_

However, the generated `minimum_ram_size` is `6164`.

ia0, 2019-11-06:
> Based on my understanding of `userland_generic.ld` in `libtock-c`, it seems like `elf2tab` assumes that the stack is allocated outside of `.data` and `.bss` section.

As far as I can tell, `userland_generic.ld` in `libtock-c` does not "allocate" the stack in the binary. Also, this doesn't seem necessary since the stack should have no content (as does `.bss`). Only its size matters (which as far as I can tell is stored in `.crt0_header` and passed as an argument to `elf2tab`).

Also, it looks like the `AT (_etext)` is not working (which means that we consume space in the binary for the flash). Here is what I see for the blink example:
```
  [ 3] .stack            PROGBITS        20002000 002000 000800 00  AX  0   0  1
  [ 4] .data             PROGBITS        20002800 002800 000000 00  WA  0   0  1
```

Are those problems new? Do we know how to fix them? Could we remove the `.stack` section and rely on `.crt0_header` and `elf2tab --stack`? I'm not familiar enough to answer those questions.

jrvanwhy, 2019-11-07:
There have been some changes since Rajiv opened this issue; I don't know if the erroneous RAM size calculation is still an issue.

Unlike libtock-c, libtock-rs' layout file needs to know the stack size so it can place .data correctly (in RAM). I agree it would probably be better to avoid placing the stack in the .data section, however, as it is ostensibly a separate allocation. If we're needlessly adding 2 KiB to the application image size then we definitely should fix it.

ia0, 2019-11-07:
Regarding erroneous RAM size calculation, there is still an issue (but I think is unrelated to the stack issue). `elf2tab` assumes that the first segment that is `LOAD RW` is the RAM segment: https://github.com/tock/elf2tab/blob/6bf7de01327a12c700b328dd2990a9765ffb5286/src/main.rs#L101-L102. This is an issue if the program uses `static mut` to define a writeable flash region since the `.wfr` section is gonna be `LOAD RW` and not merged into the `LOAD RE` segment containing `.text`. Then `elf2tab` is gonna think it's the RAM and compute the wrong size. A workaround is to use `static` such that the `.wfr` section ends up in a `LOAD RWE` segment with `.text` (and `.crt0_header`).

Regarding the stack issue, if I understand correctly, because Rust doesn't support relocations we have to place `.data` at a precise address. Because of that we have to make sure the relocations are correctly applied during linking by taking into account the stack size. Makes sense, thanks for the info!

jrvanwhy, 2019-11-08:
In practice, Rust has enough mutable state that we need to support it. I certainly don't think that "normal" Rust code (code that is sound as per Rust's normal semantics) should result in memory unsafe behavior in a Tock application, so you would need to limit everything (libtock-rs and applications) to the subset of the language that doesn't use `static mut`.

Writing to flash at runtime isn't an option for Google's use cases, as our firmware images are signed.

ia0, 2019-11-08:
I'm not sure I understand what you mean. What "normal" code would result in unsafe behavior?

Regarding flash, how do you implement persistent storage if you can't write to flash? There's a semantics difference between `static mut` in RAM and `static mut` in flash. Regarding soundness, both require `unsafe` so the responsibility of checking soundness is on the application (not Tock). The workaround of using `static` is to overcome a limitation of `elf2tab` and is indeed ugly and temporary but a pragmatic solution.

jrvanwhy, 2019-11-08:
> I'm not sure I understand what you mean. What "normal" code would result in unsafe behavior?

I think I misunderstood you -- are you suggesting trying to simplify the relocation problem by removing support for `.data` and `.bss`, or are you suggesting something else?

> Regarding flash, how do you implement persistent storage if you can't write to flash?

Mutable flash pages are separate from executable flash pages on our platform.

ia0, 2019-11-08:
> I think I misunderstood you -- are you suggesting trying to simplify the relocation problem by removing support for `.data` and `.bss`, or are you suggesting something else?

Yes, I'm suggesting that. Although we still need 20 bytes of `.bss` for the allocator (on cortex). So we still need to relocate those addresses.

> Mutable flash pages are separate from executable flash pages on our platform.

I see. But you would still need some unsafe code to access those mutable flash pages. I'm not sure what "Writing to flash at runtime isn't an option" meant and to what it referred.

jrvanwhy, 2019-11-08:
> Yes, I'm suggesting that. Although we still need 20 bytes of `.bss` for the allocator (on cortex). So we still need to relocate those addresses.

In that case, how would we support the following Rust app?

```
static mut x: u32 = 0;

fn print(value: u32) {
    // Source emitted, but equivalent to:
    // println!("{}", value);
}

fn main() {
    unsafe {
        x = 3;
        print(x);
    }
}
```

This is a sound Rust program that prints 3.

> > Mutable flash pages are separate from executable flash pages on our platform.
> 
> I see. But you would still need some unsafe code to access those mutable flash pages. I'm not sure what "Writing to flash at runtime isn't an option" meant and to what it referred.

The mutable flash pages are accessed through system call APIs, and have to be separate from the pages that contain the application image.

ia0, 2019-11-08:
> In that case, how would we support the following Rust app?

This would not be supported since it has a global mutable state. This simplification only holds for programs that don't have a global mutable state (in RAM) and should not replace the current solution. It's an alternative for programs that can't afford having the stack in the binary. But as I said in my first message, I don't think this optimization is worth it, otherwise people would already have complained about their Tock binary size.

> The mutable flash pages are accessed through system call APIs, and have to be separate from the pages that contain the application image.

Is this separation done when linking the binary or at a later stage? In particular are the mutable flash pages defined as writeable flash regions in the Tock binary (see https://github.com/tock/tock/blob/master/doc/TockBinaryFormat.md#2-writeable-flash-region) or do they use another mechanism? Having two different flash regions (one mutable and one for execution) looks interesting and I'm not sure I've seen support in Tock for it. What I've seen is that the application flash is read-execute only.

jrvanwhy, 2019-11-08:
> Is this separation done when linking the binary or at a later stage? In particular are the mutable flash pages defined as writeable flash regions in the Tock binary (see [...]

At the moment, we don't have any per-application mutable pages in flash. The few mutable pages we have are global, and owned by the kernel. Correcting my earlier statement, I'm not sure whether we can support Tock's per-application mutable flash regions (it depends on details of our code signing setup that I'm not familiar with).

jrvanwhy, 2022-02-09:
The linker script was rewritten for the Tock 2.0 crates, so I do not believe this issue is relevant anymore.

closed 2022-02-09 by jrvanwhy