tock/libtock-rs #84: #[derive(Debug)] seems expensive

jrvanwhy, 2019-04-23:
The code generated by `#[derive(Debug)]` uses `core::fmt::DebugStruct` and its siblings: `DebugList`, `DebugMap`, `DebugSet`, and `DebugTuple`. Internally, these types use `&dyn fmt::Debug`, and it appears that in Rust userspace apps this defeats LLVM's devirtualization. This causes most or all of the `Debug` implementations (including `Debug` implementations for types from libcore) to be compiled in.

I ran some tests on a stripped-down libtock-rs that removes all of its `#[derive(Debug)]` instances. In these tests, I manually derived Debug using the core::fmt::Debug* types as well as implemented it entirely manually. Here are the resulting sizes of .text (in bytes):

1. DebugStruct: 15664
1. DebugTuple: 15504
1. DebugList: 15584
1. DebugSet: 15584
1. DebugMap: 16576
1. Manual implementation (no Debug*): 504

Based on this test, I think it may be wise to avoid `#[derive(Debug)]` in libtock-rs and instead derive Debug manually. I count 12 uses of `#[derive(Debug)]` in libtock-rs so this seems reasonable.

Any comments before I make this change?

gendx, 2019-08-26:
This could also be gated by one (or several) feature(s). `cfg_attr` can be used to conditionally derive Debug:

```rust
#[cfg_attr(feature = "derive_debug", derive(Debug))]
struct Foo { ... }
```

And the release/debug mode (or another feature) could be checked to decide whether to implement Debug at all (whether hand-rolled or derived).

What do you think?

torfmaster, 2019-11-13:
I personally would be okay with just removing all of the debug implementation as I don't use them for debugging and we shouldn't use them in production code at all.

jrvanwhy, 2019-11-13:
> I personally would be okay with just removing all of the debug implementation as I don't use them for debugging and we shouldn't use them in production code at all.

This has been affecting our test binaries significantly, where we need Debug or similar functionality. Note that this also applies to Display, although I'm not sure how necessary that is in production builds.

I've been working on code size tooling to help understand how we can mitigate this issue.

gendx, 2019-11-18:
I just found out that we are not the only ones wanting a more lightweight formatting in Rust: https://twitter.com/japaric_io/status/1196203591737991168

There is now the `ufmt` crate, streamlining a lightweight `Debug`-like trait: https://crates.io/crates/ufmt

alistair23, 2022-09-14:
After talking to some Rust people they pointed me to https://github.com/rust-lang/rust/pull/101568 which should help here.

It should allow more optimisations when using the core `fmt`

m-ou-se, 2022-09-14:
> [..] this defeats LLVM's devirtualization.

Working on it!

(No promises yet, but it looks like at least one of my experimental approaches doesn't break devirtualization.)

owenthewizard, 2026-04-17:
It's been a while... Have you considered seeking maintainership of ufmt?