tock/libtock-rs #93: Hail development board bluetooth issue

ibiBgOR, 2019-10-06:
I'm using the [Hail development board](https://www.tockos.org/blog/2017/introducing-hail/) and trying to setup bluetooth. I tried the example simple_ble to broadcast values.
As far as I understand the code, the BleAdvertisingDriver sets up the chip as an advertiser ["service"](https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/ble-services-a-beginners-tutorial), am I correct?

Well my problem is, that I can't discover the chip and the data it sends. After adding some console statements I detected that the call to `BleAdvertisingDriver::initialize` leads to an error. The resulting error code is `-11`. Where do I find some documentation about what this error code is trying to tell me? Does it work on your end? Do you have some hints, how I can get BLE up and running? The code I'm using is (as mentioned above) the simple_ble example. 

Would be nice if you could help me.

P.S.: Is the ble_scanning example for discovery of other bluetooth services? Or what is this example showing?

brghena, 2019-10-06:
Unfortunately, the issue you're running into is how BLE is supported on different boards. The short version of the answer is that BLE on Hail isn't supported in the Rust userland.

On the nRF52DK and nRF52840DK, Tock runs right on the nrf microcontroller. So the BleAdvertisingDriver uses [a capsule](https://github.com/tock/tock/blob/59f8092dbfbf4f218bdf4b82df28b1d5b8b16517/capsules/src/ble_advertising_driver.rs) that uses the radio hardware to directly advertise.

On Hail, Tock runs on the SAM4L microcontroller (which has no radio) and communicates with the nRF51822 over a UART channel. To support this, we reuse the serialization libraries written in C by Nordic, however that's in [libtock-c](https://github.com/tock/libtock-c/tree/master/libnrfserialization). Porting all of that serialization code to Rust would be a good bit of work, and honestly probably not that valuable since we'd rather just support the nRF52 which have radios built right in.

As for the error code, based on the [list of error codes](https://github.com/tock/tock/blob/master/kernel/src/returncode.rs) you were receiving ENODEVICE, which occurred because you were trying to make a call to a driver in the kernel that doesn't exist for the Hail board (the BLE advertising driver).

brghena, 2019-10-06:
Also, for some extra information on BLE: 

The simple_ble app starts [BLE advertisements](https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/ble-advertising-a-beginners-tutorial) which are transmitted every [100 ms](https://github.com/tock/libtock-rs/blob/master/examples/simple_ble.rs#L43) and include in their payload:

 * A flag
 * An "incomplete list of 16-bit service UUIDs"  i.e. some service it supports
 * A "complete local name" i.e. device name "Tock"
 * A "service data" field, which can include service-specific data (in this case a serialized struct)

see [payload possible types here](https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile/). I'm not entirely sure of the origin of the particular UUIDs used in this app.

The BLE scanning app receives advertisements, looks for that particular service data field, reads out the payload from it back into a struct, and then sets LEDs based on it. So if you set up two boards, one with each app, they would be communicating over BLE advertisements.

To get services involved more than just as payload descriptions, you need to first form a BLE connection, which I believe Tock doesn't have support for yet. For example, all the information [about GATT and characteristics](https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/ble-services-a-beginners-tutorial) only applies for two devices that are in a BLE connection.

However, in libtock-c for Hail, since we use serialization to send controls to the nRF51 (which just runs a BLE stack written by nordic in embedded C on it), connections are supported. So you could connect to a Hail with your phone, or use it to connect to some other device. Unfortunately, you have to do it in the C userland. : (

torfmaster, 2019-11-13:
Did I get the conversation right that the reason for the error lies outside `libtock-rs`? If so can I close the issue?

brghena, 2019-11-13:
Yeah, thanks for paying attention to old issues @torfmaster. This is good to close for now and if there are other problems with BLE we can re-open it or start a new issue.

closed 2019-11-13 by brghena