tock/libtock-rs #106: Support for RISC-V without atomic

alistair23, 2019-11-20:
Trying to compile libtock-rs applications for platforms without atomics (OpenTitan) fails:

You can try by running this command

```shell
cargo build --release --target=riscv32imc-unknown-none-elf --example hello
```

and you get this error

```shell
error[E0599]: no method named `compare_and_swap` found for type `core::sync::atomic::AtomicBool` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/mutex.rs:131:25
    |
131 |         while self.lock.compare_and_swap(false, true, Ordering::Acquire) != false
    |                         ^^^^^^^^^^^^^^^^ method not found in `core::sync::atomic::AtomicBool`

error[E0599]: no method named `compare_and_swap` found for type `core::sync::atomic::AtomicBool` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/mutex.rs:181:22
    |
181 |         if self.lock.compare_and_swap(false, true, Ordering::Acquire) == false
    |                      ^^^^^^^^^^^^^^^^ method not found in `core::sync::atomic::AtomicBool`

error[E0599]: no method named `fetch_add` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:199:31
    |
199 |         let value = self.lock.fetch_add(READER, Ordering::Acquire);
    |                               ^^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_sub` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:205:23
    |
205 |             self.lock.fetch_sub(READER, Ordering::Release);
    |                       ^^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_sub` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:224:19
    |
224 |         self.lock.fetch_sub(READER, Ordering::Release);
    |                   ^^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_and` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:236:19
    |
236 |         self.lock.fetch_and(!(WRITER | UPGRADED), Ordering::Release);
    |                   ^^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_or` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:328:22
    |
328 |         if self.lock.fetch_or(UPGRADED, Ordering::Acquire) & (WRITER | UPGRADED) == 0 {
    |                      ^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_add` found for type `&'rwlock core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:440:19
    |
440 |         self.lock.fetch_add(READER, Ordering::Acquire);
    |                   ^^^^^^^^^ method not found in `&'rwlock core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_add` found for type `&'rwlock core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:467:19
    |
467 |         self.lock.fetch_add(READER, Ordering::Acquire);
    |                   ^^^^^^^^^ method not found in `&'rwlock core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_sub` found for type `&'rwlock core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:511:19
    |
511 |         self.lock.fetch_sub(READER, Ordering::Release);
    |                   ^^^^^^^^^ method not found in `&'rwlock core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_sub` found for type `&'rwlock core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:521:19
    |
521 |         self.lock.fetch_sub(UPGRADED, Ordering::AcqRel);
    |                   ^^^^^^^^^ method not found in `&'rwlock core::sync::atomic::AtomicUsize`

error[E0599]: no method named `fetch_and` found for type `&'rwlock core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:531:19
    |
531 |         self.lock.fetch_and(!(WRITER | UPGRADED), Ordering::Release);
    |                   ^^^^^^^^^ method not found in `&'rwlock core::sync::atomic::AtomicUsize`

error[E0599]: no method named `compare_exchange` found for type `&core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:545:16
    |
545 |         atomic.compare_exchange(current, new, success, failure)
    |                ^^^^^^^^^^^^^^^^ method not found in `&core::sync::atomic::AtomicUsize`

error[E0599]: no method named `compare_exchange_weak` found for type `&core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/rw_lock.rs:547:16
    |
547 |         atomic.compare_exchange_weak(current, new, success, failure)
    |                ^^^^^^^^^^^^^^^^^^^^^ method not found in `&core::sync::atomic::AtomicUsize`

error[E0599]: no method named `compare_and_swap` found for type `core::sync::atomic::AtomicUsize` in the current scope
   --> /home/alistair/.cargo/registry/src/github.com-1ecc6299db9ec823/spin-0.5.2/src/once.rs:104:33
    |
104 |             status = self.state.compare_and_swap(INCOMPLETE,
    |                                 ^^^^^^^^^^^^^^^^ method not found in `core::sync::atomic::AtomicUsize`

error: aborting due to 15 previous errors

For more information about this error, try `rustc --explain E0599`.
error: Could not compile `spin`.
warning: build failed, waiting for other jobs to finish...
error: build failed

```

Woyten, 2019-11-20:
Although I am not surprised that it does not work, I cannot tell you what the exact reason is. It might be the `#![no_std]` flag or the situation that `riscv32` is not a first-tier target.

Anyway, I am feeling that you are fighting against the philosophy of Tock. A `libtock-rs` process runs cooperatively with itself. It is possible to use concurrency by placing yield points (`syscalls::yieldk` or `.await`) in the code. Parallelism, however, is not possible within one process, so there shouldn't be any need for atomicity.

jrvanwhy, 2019-11-20:
> Although I am not surprised that it does not work, I cannot tell you what the exact reason is.

It is because our global memory allocation implementation relies on atomics (note we pull in the `spin` crate via the `linked-list-allocator` crate.

> Parallelism, however, is not possible within one process, so there shouldn't be any need for atomics.

Rust as a language doesn't know this, and requires that `static` objects be `Sync`. Atomics are used here to satisfy this requirement.

Woyten, 2019-11-20:
@jrvanwhy Yeah, you're right. It should be the allocator. I was understanding the problem in a wrong way.  I was working on making allocation an optional feature but I got some strange messages...

alistair23, 2019-11-21:
Any ideas how to disable that? It would be nice to run without atomics

rajivr, 2019-11-21:
@alistair23 ARMv6-M (Cortex M0) also does not have atomics. I recall running into something similar in the past when working on M0. While I cannot verify it right away, I believe you can turn atomics off by importing `linked-list-allocator` without default features.

```
[dependencies.linked_list_allocator]
path = "../linked-list-allocator"
default-features = false
```

Woyten, 2019-11-21:
I am working on a solution.

Woyten, 2019-11-21:
#107 should do the trick.

PR #107 (2019-11-22): Support for RISC-V without atomic
Fixes #106 

- The `spin` feature of the `linked_list_allocator` has been disabled, s.t. the allocator does not rely on atomics anymore.
closed 2019-11-22 by torfmaster