tock/libtock-rs #129: allow() is unsafe

jrvanwhy, 2020-01-07:
`SharedMemory` instances returned by `allow()` contain a reference to a buffer that may be mutated by the kernel. This is undefined behavior in Rust.

It's probably not possible to expose a safe API that allows an application to access an `allow()`-ed buffer until that buffer has been taken back from the kernel (i.e. overwritten with another `allow()` call).

torfmaster, 2020-01-07:
I'm not sure about that. The buffer is private and is never exposed directly. Instead, the only way to obtain data from the buffer is to copy data from/to it. When `SharedMemory` goes out of scope the kernel is told to not use the memory anymore (this is a change in the kernel we introduced when we created `SharedMemory`).

jrvanwhy, 2020-01-07:
It is undefined behavior whenever all of the following occur simultaneously:

1. `&mut [T]` exists
2. One of the `T` values in the slice is mutated through a mechanism other than the slice reference.

In this case, `SharedMemory` satisfies 1 and the kernel mutates it through an independent reference.

torfmaster, 2020-01-07:
Maybe I am missing a point here, but from my naive point of view this would imply that MMIO and DMA is undefined behavior in general. Maybe you need some non-atomicity assumption here?

jrvanwhy, 2020-01-07:
If I'm reading the nomicon correctly, MMIO works if you use shared references to types containing UnsafeCell (and use volatile reads/writes).

Woyten, 2020-01-07:
Yes, it would probably more correct to store an `UnsafeCell` or a `*mut` internally. But this can be achieved without declaring `allow` to be `unsafe`.

jrvanwhy, 2020-01-07:
I'm not suggesting we mark `allow` unsafe, but I think we need to change its API to a sound API.

jrvanwhy, 2020-01-08:
Ah, I see I put "unsafe" rather than "unsound" in the title. It should have been "unsound".

jrvanwhy, 2020-06-09:
This issue is actually worse than my original description states: it is not currently possible to un-allow a buffer shared with the kernel. See https://github.com/tock/tock/pull/1831 for background.

Note that Tock's design here may change in the future to make un-allow work reliably: https://github.com/tock/tock/issues/1905

jrvanwhy, 2022-02-09:
Solved in the Tock 2.0 ABI design and `libtock-rs` crates.

closed 2022-02-09 by jrvanwhy