tock/libtock-rs #133: [RFC] What should happen when an app panics?

jrvanwhy, 2020-01-15:
The Tock kernel does not give applications a way to stop executing. The `yield` system call continues to run callbacks, and is the only way for an app to voluntarily yield the CPU. I'm not sure that continuing to run callbacks is the desired behavior for the Rust runtime on Tock.

Here are some possibilities:

1. Keep running callbacks. This is weird but is the lightest-weight solution.
2. Add a `bool` global indicating whether the app is panicing. Modify the `subscribe` API to hide the `userdata` parameter; bounce callbacks through a function that skips the callback if the app is panicing (using `userdata` to identify the real function). The extra function and global only cost a few words but this would prevent apps from making use of the userdata parameter.
3. Ban `panic()`, by making it call an undeclared symbol. I expect this option to be the most annoying to use, but it has the advantage of eliminating Rust's panic-related formatting code.

Note that a similar issues with `main()`: what if `main` returns? However, there's an additional option here: `libtock-rs` can prevent `main()` from returning by requiring it to return an empty type.

torfmaster, 2020-01-16:
I think there should be a way for an app to stop/panic, maybe an additional syscall should take care of this (disabling callbacks or stopping the app). The kernel just shouldn't call any callbacks anymore.

torfmaster, 2020-02-23:
I created a PR in the kernel implementing the solution of adding a syscall to panic:

https://github.com/tock/tock/pull/1624

jrvanwhy, 2020-06-09:
I expect Tock 2.0 to support an exit system call, but in the meantime I think it's best to let callbacks keep running. That is the lightest-weight option that still allows `panic()`.

closed 2020-06-09 by jrvanwhy