tock/libtock-rs #134: building against riscv32imc results in unloadable binary

bcantrill, 2020-01-18:
Building against riscv32imc (e.g., for OpenTitan) does not pull in the requisite linker file, thereby resulting in a binary that does not properly have a `.crt0_header`. Upon execution, `_start` attempts to treat the beginning of program text as a `LayoutHeader`, which of course it isn't -- resulting in horrible program death.  At the very least, `.cargo/config` must be updated for riscv32imc:

```difff
diff --git a/.cargo/config b/.cargo/config
index 818ce68..64472e4 100644
--- a/.cargo/config
+++ b/.cargo/config
@@ -6,6 +6,13 @@ rustflags = [
   "-D", "warnings",
 ]
 
+[target.riscv32imc-unknown-none-elf]
+rustflags = [
+  "-C", "link-arg=-Triscv32_layout.ld",
+  "-C", "relocation-model=static",
+  "-D", "warnings",
+]
+
 [target.riscv32imac-unknown-none-elf]
 rustflags = [
   "-C", "link-arg=-Triscv32_layout.ld",
```

If riscv32imac is not to be supported at all (which is the implication of `build_examples.sh`), the name of the target should clearly just be changed.  Happy to open a pull request on either change...

torfmaster, 2020-01-18:
Thank you for reporting this. We currently already have the issue where we have no 1:1 mapping between boards and llvm targets we just cannot build binaries "out of the box". I think the solution would be (also regarding upcoming changes for slots for apps) to make the build infrastructure more flexible to be able to build an app per slot and per board.

Woyten, 2020-01-18:
I think it's a mistake that those lines are missing. You could open a PR adding those line. Besides that, the `build_examples.sh` script does not really tell you whether we support a specific target or not. It only makes sure that the code can be built with different compile time settings (`#[cfg(...)]`) and under a minimal set of assumptions (e.g. the code should not rely on atomic operations).

closed 2020-01-21 by bors[bot]