tock/libtock-rs #138: Build failure

lu-zero, 2020-02-07:
```
cargo -v build --target riscv32imac-unknown-none-elf --example blink
```
```
Compiling libtock v0.2.0 (/Users/lu_zero/Sources/rust/libtock-rs)
     Running `rustc --edition=2018 --crate-name blink examples/blink.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C panic=abort -C lto -C debuginfo=2 -C metadata=8d8e1bbf1e206f9c -C extra-filename=-8d8e1bbf1e206f9c --out-dir /Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/examples --target riscv32imac-unknown-none-elf -C incremental=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/incremental -L dependency=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/Users/lu_zero/Sources/rust/libtock-rs/target/debug/deps --extern core=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/libasync_support-0be3d2654d74ceb4.rlib --extern corepack=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/libcorepack-d96b372b0e9f5d2b.rlib --extern futures=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/libfutures-aad3dc6f7e1b0792.rlib --extern libtock=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/liblibtock-9bb9c6091e8d5c84.rlib --extern libtock_codegen=/Users/lu_zero/Sources/rust/libtock-rs/target/debug/deps/liblibtock_codegen-b3113715052c2373.dylib --extern linked_list_allocator=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/liblinked_list_allocator-00203b4dbedd8f42.rlib --extern serde=/Users/lu_zero/Sources/rust/libtock-rs/target/riscv32imac-unknown-none-elf/debug/deps/libserde-6b545f6835fcafdc.rlib -C link-arg=-Triscv32_layout.ld -C relocation-model=static -D warnings`
warning: Invalid record (Producer: 'LLVM9.0.0-rust-1.40.0-nightly' Reader: 'LLVM 9.0.0-rust-1.40.0-nightly')

error: failed to load bc of "17dodecjo1334gq8":

error: aborting due to previous error

error: could not compile `libtock`.
```

This after pinning `linked_list_allocator` to `0.6.4`

gkelly, 2020-02-07:
Huh that looks like an LLVM bug.

Woyten, 2020-02-07:
Could you try this fix and report if it helped?
https://github.com/torfmaster/libtock-rs/commit/67fc4908f7da496423ce13fb6816a8f07816e860

lu-zero, 2020-02-07:
It worked fine.

torfmaster, 2020-02-19:
Is the fix something we want to bring to master?

alistair23, 2020-02-19:
Yes please!

Woyten, 2020-02-19:
I prepared everything and could create a PR very soon.

commit 14451c4b (2020-02-19): Merge #149

149: Update Rust toolchain r=Woyten a=Woyten

Fixes #138 and updates the Rust toolchain s.t. it is in line with Tock OS.
commit 0ff9f339 (2020-02-19): Merge #149

149: Update Rust toolchain r=Woyten a=Woyten

Fixes #138 and updates the Rust toolchain s.t. it is in line with Tock OS.
PR #149 (2020-02-19): Update Rust toolchain
Fixes #138 and updates the Rust toolchain s.t. it is in line with Tock OS.
closed 2020-02-19 by bors[bot]