tock/libtock-rs #160: libtock-rs is considered messy

Woyten, 2020-03-09:
According to https://github.com/tock/tock/pull/1637/files `libtock-rs` is considered being messy. I would like to use this issue to track why this is the case and what we can do about it without needing to rewrite everything.

First of all, it is very clear that `libtock-rs` isn't a perfect library. When @torfmaster and I started to work on this library, it was in a pure proof-of-concept state with certain language features not working, with no toolchain, no tests, no code style guidelines, a C-like API, etc. The user experience was really terrible. Despite all, we liked the idea of `libtock-rs` and put a lot of effort into fixing all those issues. With the exception of the relocaton problem (#64) which was driven by @jrvanwhy we got close to zero input from other people regarding the API design decisions we made. We even tried to encourage people to work on a design document (#20) without any success.

In total, most contributions to this library from other people are limited to minor technical fixes to make their own problem disappear. Really break-through PRs (e.g. the QEMU test suite #157) are a rare exception. Without visions and bigger contributions from other people it is no suprise that this library hasn't reached a high-quality grade. Indeed, it is hard enough to retain the current quality as people are trying to add gadgets or cool examples to the library before the bigger structures have been stabilized. This is absolutely no critique against those people (Every contribution is welcome!) but it demonstrates how, currently, there is no clear focus on the important parts.

To sum it up, I feel a bit sad about the fact that help was not really provided although the possibilities were given and that, now, a discussion about a rewrite is popping up. To solve this problem constructively, we should collect issues (What is messy? / What should be removed/changed/added?) and organize them in a project. torfmaster and I have collected a lot of experience when refactoring this library. We know that incremental changes of the library are easily doable and that `libtock-rs` has the perfect size for it. On the one hand, it is big enough to evaluate the effect of an incremental change. On the other hand, it is small enough s.t. incremental changes can be done with reasonable effort.

jrvanwhy, 2020-03-09:
My primary concerns are as follows:

1. **Code size:** `libtock-rs` currently generates excessively-large applications, which is limiting (#156). I think we can improve it significantly, but there's a limit to how much we can minimize the size of apps that depend on futures. Based on the experimentation I've done and documented at [design-explorations/size_comparison](https://github.com/tock/design-explorations/tree/master/size_comparison), incremental optimization will never reduce the cost of futures to an amount acceptable to our `libtock-rs` applications
1. **Certifiability:** Applications built on `libtock-rs` pull in a lot of complex, difficult-to-understand futures code. This makes security auditing difficult.
1. **Complexity and bugginess related to futures:** Adding futures support added bugs due to the increased complexity (#113). Maintaining `tokio` on Linux is already a full-time job. Maintaining futures support on `Fuchsia` is also a full time job -- I know because I've spoken with `Fuchsia`'s futures maintainer. I don't think *we* have enough full-time contributors to maintain a high-quality futures and executor implementation.
1. **Lack of testing:** I would prefer to have much more thorough testing of the library, both at the unit test level and the integration test level.
1. **Minimal documentation:** The library is not well documented. A lot of general design points (such as the treatment of `yield` as unsafe) are not clearly documented and only come up during code review. This also leads to soundness issue, such as #129. `libtock-rs`'s architecture is nontrivial to decode.

I think concerns 4 and 5 can be addressed through incremental improvements. I think that addressing concern 4 will resolve my concerns about buginess related to futures (part of concern 3), but the high complexity of `libtock-rs` will still hurt developer velocity.

I do not believe that concern 1 and 2 can be addressed without significantly restructuring `libtock-rs`. At worst, `libtock-rs` should be usable in a manner where its code size overhead is similar to that of Tock's kernel. Tock's kernel is generally lean, except it uses dynamic polymorphism for client callbacks in many places -- which brings in vtables (each vtable in Rust wastes 2 words too!) and inhibits optimidzation. I have an idea on how to do better, but I have not *demonstrated* that idea. Because I do not have a concrete design yet I haven't put forth a proposal on what `libtock-rs` should look like.

I think `libtock-rs` should be structured in two layers. The lower-level API for the library would be as low-cost as possible (minimal unnecessary code bloat, RAM usage, or cycles wasted). The higher-level API in `libtock-rs` would be futures-based. Applications that care about code size and certification would only use the low-level API, while other applications would be free to use the futures-based API.

The problem is if we create two APIs in isolation from each other, then applications authors have a difficult choice to make up-front. Using the low-level API from the start would hurt productivity for the application's authors. However, using the futures-based API from the start would risk forcing the application's authors to completely rewrite their app if they later run out of flash/RAM space.

Therefore, there needs to be a reasonable path for applications to migrate from the futures-based API to the low-level API. Then application developers can start writing apps using futures and migrate incrementally to the low-level API if they run into size constraints.

With that in mind, here's what I've been working on at the moment:

1. Create a prototype `libtock-rs` that uses a more efficient coding structure than Tock's kernel (no futures, no unnecessary vtables, no unnecessary dynamic polymorphism). Port [OpenSK](https://www.github.com/google/opensk) to my `libtock-rs` prototype to prove it is usable.
1. Build a futures-based API on top of the lower-level API.
1. Port OpenSK to the futures-based API then incrementally migrate it to the lower-level API to demonstrate that incremental migration is possible.

Woyten, 2020-03-09:
@jrvanwhy Thanks for your feedback. This is only a quick response. I will probably write some more text once I find the time for it.

My impression is that most of your concerns are related to futures. I do not consider them as something you *must* use in any case but as something a modern library *should* offer. If we throw away the `block_on` code in the `panic_handler`/`alloc_error_handler` and offer a yielding `sleep_unsafe` method, then it's completely up to the user whether he wants to use futures or not. Would this be a solution? It might not even be a breaking change.

alistair23, 2020-03-09:
I generally agree with @Woyten's statements about the issues with `libtock-rs`. I feel it's heavy and complex. I'm not sure it should be re-written though, instead I would like to see improvements made incrementally. If that ends up re-writing everything then that is fine, but a clean start seems like too much. There is still a lot of good things here.

I do like the layer approach though. It would be nice if `libtock-rs` was just entry code and wrappers around the syscalls (as small as possible), but included features that could be enabled. So if you want futures you can enable futures (which is supported in `libtock-rs`).

jrvanwhy, 2020-03-09:
> My impression is that most of your concerns are related to futures. I do not consider them as something you _must_ use in any case but as something a modern library _should_ offer. If we throw away [...]

Whether that is a solution depends on what we want `libtock-rs` to handle. Should `libtock-rs` just be the Rust runtime (e.g. entry point) and syscalls (command, yield, allow, etc...), or should `libtock-rs` contain higher-level abstractions over each of the syscall APIs (e.g. a timer, a GPIO abstraction, buttons, etc).?

If `libtock-rs` is limited to the runtime and syscalls, then its futures support should be minimal but applications would need to interface with the syscall APIs themselves. If `libtock-rs` is responsible for providing higher-level abstractions then we should provide the same APIs in the low-cost and the futures-based layers.

Woyten, 2020-03-09:
Could you explain why you think `libtock-rs` is complex? I am aware that futures are considered complex and I kind of agree. But what about the remaining code and idioms?

jrvanwhy, 2020-03-09:
Are you asking Alistair or me? I think there are 2 sources of `libtock-rs`'s complexity:

1. Unavoidable complexity because `libtock-rs` is a language runtime and because `libtock-rs` provides FFI in the form of the system call interface.
1. Complexity due to the futures API.

alistair23, 2020-03-10:
> Could you explain why you think `libtock-rs` is complex? I am aware that futures are considered complex and I kind of agree. But what about the remaining code and idioms?

I think the best answer to what I mean is the questions that I have asked myself in the past :) Mostly related to Futures though.
 - What is codegen?
 - Why is there a main in codegen and in my example?
 - What is this build.rs file?
 - How do I build my example for this platform?

labbott, 2020-03-10:
I'm also +1 to libtock-rs just being syscall wrappers + a minimal entry point and having a separate place for higher level abstractions including Futures.

Woyten, 2020-03-10:
So, where's the complexity of the *library* - not the toolchain - apart from the futures?

Woyten, 2020-03-10:
> I'm also +1 to libtock-rs just being syscall wrappers + a minimal entry point and having a separate place for higher level abstractions including Futures.

I agree that the split makes sense but shouldn't the high-level API keep the name `libtock-rs`? The low-level API could be named `libtock-core` or the like?

Woyten, 2020-03-10:
>     3\. Adding futures support added bugs due to the increased complexity (#113).

As far as I remember, this bug had nothing to do with futures.

torfmaster, 2020-03-11:
I think we could offer a base lib/crate  ´libtock-rs-core` just wrapping the system calls. However, I would create at least _some_ kind of convenience library containing drivers on top. When I look at the examples (which for me reflect real world applications) it would be horrible to write them just using syscalls (it would feel like writing unix applications just using syscalls and inline assembly).

I'm thinking of three layers here: ` libtock-async` -> ` libtock-rs`  -> ` libtock-core`. The api of ` libtock-async`  would be completely asynchronous. `libtock-rs` would contain most of the drivers to be used synchronously. No async main, no futures. ` libtock-core` would then contain the panic handler and the syscalls.

torfmaster, 2020-03-11:
> * What is codegen?
> * Why is there a main in codegen and in my example?
> * What is this build.rs file?

I think these questions are not fair: Firstly, they refer to implementation details in the library upon first use (I would for example not ask which clobbers in the entry point are necessary, I just use the library). Moreover, these points are documented or basic Rust knowledge.

torfmaster, 2020-03-11:
@jrvanwhy How does #113 relate to futures?

torfmaster, 2020-03-11:
> Maintaining tokio on Linux is already a full-time job. Maintaining futures support on Fuchsia is also a full time job -- I know because I've spoken with Fuchsia's futures maintainer. I don't think we have enough full-time contributors to maintain a high-quality futures and executor implementation.

I think maintaining `block_on` in `async_std` is not a full time job. Moreover, I don't expect more sophisticated executors to come.

jrvanwhy, 2020-03-11:
> I'm thinking of three layers here: ` libtock-async` -> ` libtock-rs` -> ` libtock-core`. The api of ` libtock-async` would be completely asynchronous. `libtock-rs` would contain most of the drivers [...]

I agree, except that I think the no-futures drivers (which you called `libtock-rs`) to be asynchronous. We should support apps that need an async API but which cannot accept the code size cost of futures. This related to my previous point about migration from the futures-based to the non-futures-based API.

I don't think we need a separate synchronous API *if* it is easy to use the futures in a synchronous fashion (i.e. `block_on` is easy to use everywhere).

torfmaster, 2020-03-11:
> We should support apps that need an async API but which cannot accept the code size cost of futures.

For me async is tantamount with using futures, in particular != callbacks. Sorry for the confusion. So libtock-async would use futures, libtock-rs won't (although, of course, it supports asynchronous operations using callbacks).

jrvanwhy, 2022-02-09:
Closing this issue, as `libtock-rs` was completely rewritten for Tock 2.0. While it isn't perfect now, the concerns described in this issue no longer apply.

closed 2022-02-09 by jrvanwhy