tock/libtock-rs #171: release builds are undebuggable

bcantrill, 2020-03-30:
Currently, the `release` profile builds with its default of leaving `debug` as `false` -- creating binaries that are (unsurprisingly!) undebuggable.  libtock-rs should do what Tock itself does and explicitly set this to `true` for both `profile.release` and `profile.dev`:

```diff
diff --git a/Cargo.toml b/Cargo.toml
index eb56daa..26aee41 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -51,10 +51,12 @@ required-features = ["alloc"]
 [profile.dev]
 panic = "abort"
 lto = true
+debug = true
 
 [profile.release]
 panic = "abort"
 lto = true
+debug = true
 
 [workspace]
 members = [
```

alistair23, 2020-03-30:
Won't that increase binary size though?

bcantrill, 2020-03-30:
No, the `.debug` sections are not loadable, e.g. for the `hello_world` example:

```
$ riscv32-unknown-elf-objdump -h hello_world.elf 

hello_world.elf:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .crt0_header  00000040  20030040  20030040  00001040  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .text         00000d34  20030080  20030080  00001080  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .stack        00000800  10002800  10002800  00001db4  2**0
                  ALLOC, READONLY, CODE
  3 .data         00000000  10003000  20030db4  00002000  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .got          00000000  10003000  20030db4  00002000  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  5 .bss          00000004  10003000  20030db4  00002000  2**0
                  ALLOC
  6 .endflash     00000000  20030db4  20030db4  00002db4  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  7 .debug_str    0000cfc7  00000000  00000000  00002db4  2**0
                  CONTENTS, READONLY, DEBUGGING
  8 .debug_loc    00000e29  00000000  00000000  0000fd7b  2**0
                  CONTENTS, READONLY, DEBUGGING
  9 .debug_abbrev 000007fc  00000000  00000000  00010ba4  2**0
                  CONTENTS, READONLY, DEBUGGING
 10 .debug_info   0000d6b9  00000000  00000000  000113a0  2**0
                  CONTENTS, READONLY, DEBUGGING
 11 .debug_aranges 00000708  00000000  00000000  0001ea59  2**0
                  CONTENTS, READONLY, DEBUGGING
 12 .debug_ranges 00003610  00000000  00000000  0001f161  2**0
                  CONTENTS, READONLY, DEBUGGING
 13 .debug_macinfo 00000002  00000000  00000000  00022771  2**0
                  CONTENTS, READONLY, DEBUGGING
 14 .debug_pubnames 00003065  00000000  00000000  00022773  2**0
                  CONTENTS, READONLY, DEBUGGING
 15 .debug_pubtypes 00003c7a  00000000  00000000  000257d8  2**0
                  CONTENTS, READONLY, DEBUGGING
 16 .debug_frame  00002030  00000000  00000000  00029454  2**2
                  CONTENTS, READONLY, DEBUGGING
 17 .debug_line   000074c3  00000000  00000000  0002b484  2**0
                  CONTENTS, READONLY, DEBUGGING
 18 .comment      00000012  00000000  00000000  00032947  2**0
                  CONTENTS, READONLY
```

alistair23, 2020-03-30:
Do you want to send a PR?

bcantrill, 2020-03-30:
Sure, happy to!

closed 2020-04-13 by bors[bot]