tock/libtock-rs #175: 'make hail' not flashing any apps to device

JOE1994, 2020-04-27:
Hello, 

First of all, thank you for the effort of managing this repo!

I was trying out `libtock-rs` by following the guidelines in the README (with my `hail` board)

```shell
$ make setup
$ make hail  # This should probably build and flash the examples to my board?

$ tockloader list
No device name specified. Using default "tock"
Using "/dev/ttyUSB0 - Hail IoT Module - TockOS"

No found apps.
Finished in 1.894 seconds
```

It seems that running `make hail` doesn't flash any example apps to my board..
I wanted to ask if there is something else I need to do here. Thanks!

P.S. I successfully installed a 'blinky' app to my board today (and erased it afterwards), so I guess it shouldn't be a mechanical issue with my board.

ppannuto, 2020-04-27:
From the usage:

```
$ make
Welcome to libtock-rs!

First things first, if you haven't yet, check out Tocks's doc/Getting_Started.
After that read the README from libtock-rs
You'll need to install a few requirements before we get going.

The next step is to choose a board to build Tock for. Mainline
libtock-rs currently includes support for the following platforms:
 - hail
 - nrf52840
 - opentitan
 - hifive1
 - nrf52

Run 'make setup' to setup Rust to build libtock-rs.
Run 'make <board>' to build libtock-rs for that board
    Set the DEBUG flag to enable the debug build
    Set the FEATURES flag to enable features
Run 'make flash-<board> EXAMPLE=<>' to flash EXAMPLE to that board
Run 'make test' to test any local changes you have made
```

I think you need to run `make flash-hail` if you want the program to actually load; `make hail` just builds.

ppannuto, 2020-04-27:
Maybe the README could use a PR updating its documentation? I don't track libtock-rs that carefully, so I've never actually followed the steps in the README! :)

JOE1994, 2020-04-27:
> I think you need to run `make flash-hail` if you want the program to actually load; `make hail` just builds.

That worked! Thank you :+1: I can submit a PR making small changes to the README directions later today :smiley_cat:

closed 2020-04-27 by JOE1994