tock/libtock-rs #182: Cannot run example

bradjc, 2020-05-15:
I'm trying to run blink on HiFive1b. I get:

```
$ make flash-hifive1 EXAMPLE=blink
PLATFORM=hifive1 cargo run --release --target=riscv32imac-unknown-none-elf --example blink
   Compiling libtock v0.2.0 (/Users/bradjc/git/libtock-rs)
    Finished release [optimized + debuginfo] target(s) in 1.60s
     Running `/Users/bradjc/git/libtock-rs/./tools/flash.sh target/riscv32imac-unknown-none-elf/release/examples/blink`
++ basename target/riscv32imac-unknown-none-elf/release/examples/blink
+ artifact=blink
+++ dirname target/riscv32imac-unknown-none-elf/release/examples/blink
++ readlink -f target/riscv32imac-unknown-none-elf/release/examples/../..
readlink: illegal option -- f
usage: readlink [-n] [file ...]
+ rust_target_folder=
make: *** [flash-hifive1] Error 1
```
Is this a mac issue?

ppannuto, 2020-05-15:
Yes.

BSD readlink:
```
     -f format
             Display information using the specified format.  See the FORMATS
             section for a description of valid formats.
```

Coreutils readlink:
```
       -f, --canonicalize
              canonicalize by following every symlink in every  component  of
              the  given  name  recursively;  all but the last component must
              exist
```

torfmaster, 2020-05-23:
Being platform independent could be achieved by using a rust binary for flashing instead of a shell script (which seems to be hard to make compatible with different *xes). If no one objects I would replace flash.sh using a rust binary using `std::fs::read_link` which hopefully does what it promises.

A (not preferred) alternative would be doing something like https://stackoverflow.com/questions/1055671/how-can-i-get-the-behavior-of-gnus-readlink-f-on-a-mac and hoping that it works.

ppannuto, 2020-05-26:
We've taken both approaches in the tock tree (write a little rust utility or do platform detection in shell scripts). I think either solution is fine. I personally write shell scripts, but that's because I'm faster at that, not for any other reason.

torfmaster, 2020-05-26:
One advantage of using a rust utility could be that it could use a `tockloader` crate easily once it exists. As this is a pure infrastructure task I would give it a "help needed" label.

bradjc, 2020-06-15:
Any ETA on this? Tock should support both linux and mac.

ppannuto, 2020-06-15:
We use the "perl is everywhere" solution elsewhere in Tock to work around x-platform issues. This is a oneliner that fixes: https://stackoverflow.com/a/24572274/358675

fgsch, 2020-06-15:
In the interest of moving forward and unblocking macos users, can we use:

```
-rust_target_folder="$(readlink -f $(dirname $1)/../..)"
+rust_target_folder="$(cd $(dirname $1)/../.. && pwd -P)"
```

This seems trivial and doesn't require a full <insert language interpreter>.

torfmaster, 2020-06-16:
@fgsch To unblock MacOS users I would prefer your solution. However, any solution described here shows that bash scripts don't really fit our purposes: They're hard to test and are likely to fail even on slightly exotic platforms as MacOS. Your workaround looks promising, however, this makes the scripts even harder to understand and to maintain.
So I would replace them by a tiny proper Rust binary (and create an extra issue for this).

torfmaster, 2020-06-16:
@ppannuto I am a bit biased against introducing yet another language (I don't know, I have to admit) into the Tock world. Currently, we have: cargo build system, Rust, C, python, Makefiles, shell scripts. Adding perl to this list would not really improve the situation: From my experience maintainance of the build system is not very popular among developers, having to master so many technologies doesn't make it more popular.

ppannuto, 2020-06-16:
My bias is to working development environments. perl has been "part of" [the libtock-c build system](https://github.com/tock/libtock-c/blob/880b9debe582af0418e070a682ef66a5c46f8aea/AppMakefile.mk#L198) since [2017, when libtock-c was still part of tock proper](https://github.com/tock/tock/commit/df118a653b1e5d311d4c6a8f7732e86a434e70fd).

Copying a 1-liner from stack overflow != a polyglot build system.

Mac support has apparently been broken for libtock-rs since May, that is far far worse than one line of perl.

commit b554e9ba (2020-06-16): replace `readlink`

Fixes #182.
commit 18c41f65 (2020-06-16): Merge #200

200: replace `readlink` r=jrvanwhy a=ppannuto

Fixes #182.

---

Yes, there are more elaborate fixes available, but this is an easy case where perfect is the enemy of good. Let's unblock mac users.
PR #200 (2020-06-16): replace `readlink`
Fixes #182.

---

Yes, there are more elaborate fixes available, but this is an easy case where perfect is the enemy of good. Let's unblock mac users.
closed 2020-06-16 by bors[bot]