tock/libtock-rs #209: libtock_test flaky: occasionally faults

jrvanwhy, 2020-06-24:
I can reproduce the issue with the following command sequence:

```
make test-qemu-hifive  # To build everything (*may* reproduce issue)
while [ "$?" == "0" ] ; do target/debug/test_runner ; done
```

Sample output from a failing run:

```
UART: HiFive1 initialization complete.
UART: Entering main loop.
UART: [      OK ] Console
UART: [      OK ] static mut
UART: [      OK ] Dynamic dispatch
UART: [      OK ] Formatting
UART: [      OK ] Heap
UART: [      OK ] Drivers only instantiable once
UART: [      OK ] Callbacks
UART: 
UART: Kernel panic at <::core::macros::panic macros>:5:
UART:   "Process libtock_test had a fault"
UART:   Kernel version release-1.5-213-g152189fe
UART: 
UART: ---| No debug queue found. You can set it with the DebugQueue component.
UART: 
UART: ---| RISC-V Machine State |---
Error: Elapsed(())
```

The fault occurs after the following line starts executing: https://github.com/tock/libtock-rs/blob/a76b21316d9d22b641129795e081d41293be6f69/examples-features/libtock_test.rs#L127.

The fault occurs before the following line begins printing: https://github.com/tock/libtock-rs/blob/a76b21316d9d22b641129795e081d41293be6f69/examples-features/libtock_test.rs#L33.

_Originally posted by @jrvanwhy in https://github.com/tock/libtock-rs/pull/206#issuecomment-649132981_

alistair23, 2020-06-25:
This seems to be related to the MTIMER interrupt. During the debug handler the HiFive1 overruns the stack, which is why you don't see all of the prints. I'm assuming that's why it only happens some times as well as it just depends on when the timer fires. I'm not sure why it's causing a fault though.

jrvanwhy, 2021-12-12:
I haven't seen this in a long time, I assume it was fixed (IIRC the whole Tock 2.0 project happened between when I opened this issue and now).

closed 2021-12-12 by jrvanwhy