tock/libtock-rs #216: Label cleanup

jrvanwhy, 2020-07-07:
I think that when #215 is merged we'll want to use labels to identify upkeep and significant PRs. Assigning labels requires scrolling up and down through the labels list, which currently has 17 labels (and will have 18 if we add a "significant" label). I'd like to shorten that list so it takes less time to add labels to PRs and issues.

I propose deleting the following labels, which have never been used in `libtock-rs`:

1. [blocked](https://github.com/tock/libtock-rs/issues?q=label%3Ablocked)
1. [duplicate](https://github.com/tock/libtock-rs/issues?q=label%3Aduplicate)
1. [invalid](https://github.com/tock/libtock-rs/issues?q=label%3Ainvalid)
1. [wontfix](https://github.com/tock/libtock-rs/issues?q=label%3Awontfix)

Let me know if you agree or disagree!

alistair23, 2020-07-08:
Why have an invalid tag? Why not just close the PR instead? The same goes with duplicate.

I like `bug` and `feature` for issues and `significant` and `encahancement` for PRs

jrvanwhy, 2020-07-08:
> Why have an invalid tag? Why not just close the PR instead? The same goes with duplicate.

I think the `invalid` and `duplicate` labels are for issues, not for PRs. Regardless, we have never used them in libtock-rs.

alistair23, 2020-07-09:
Ah, I misread it and thought you wanted to keep only those four.

Yep, they should be deleted.

jrvanwhy, 2020-07-14:
I deleted the blocked, duplicate, invalid, and wontfix labels.

closed 2020-07-14 by jrvanwhy