tock/libtock-rs #219: How to test bluetooth examples ('simple_ble', 'ble_scanning') ?

JOE1994, 2020-07-09:
Hello :crab: ,

may I ask how to test the bluetooth example programs??
I used the Android app [nRF Connect for Mobile](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) on my phone for scanning BLE signals and advertising BLE signals.

* `simple_ble`
This one seems to perform BLE advertising, but the nRF app fails to detect any nearby signal with a display name of **"Tock!"**.

* `ble_scanning`
I activated an BLE advertiser from the nRF app, but my `hail` board doesn't respond to it.. (LED is not turned on)

Thank you! :crab:

torfmaster, 2020-07-09:
The bluetooth examples work on nordic boards. However, the hail kernel does not have the required drivers, so the apps cannot work. If you don't mind I would close this issue as it is not an issue of `libtock-rs`.

JOE1994, 2020-07-09:
Thank you for your feedback! I'll close the issue now.

closed 2020-07-09 by JOE1994