tock/libtock-rs #225: `make setup-qemu` and CI are broken

jrvanwhy, 2020-07-30:
Running `make setup-qemu` in preparation for `make test` gives the following error:

```
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information
```

The fix is to update to a Tock submodule that includes https://github.com/tock/tock/pull/2024. Unfortunately, when I do so, the tests hang in QEMU. Updating to a version that includes https://github.com/tock/tock/pull/2033 indicates that an out-of-bounds write is happening. I tried using the low level debug capsule from within libtock-rs to debug the issue, and it *appears* to me that the hang is happening before execution starts.

Notably, `libtock-rs`' Travis build passed on https://github.com/tock/libtock-rs/pull/218, and we haven't touched CI infrastructure since, so I'm not 100% sure what changed (perhaps the download URL passed back from Azure?)

@alistair23 , is this something you can help with? I'm not sure I have time to debug HiFive kernel issues.

jrvanwhy, 2020-08-03:
I think I have it working, I'll send a PR shortly.

closed 2020-08-04 by bors[bot]

alistair23, 2020-08-04:
I'm on holidays at the moment, thanks for fixing this

commit 86c36900 (2020-08-04): Merge #226

226: Fix the broken Travis CI: r=jrvanwhy a=jrvanwhy

1. Update the `tock` submodule to a version that includes
   https://github.com/tock/tock/pull/2024, which fixes the OpenTitan BootROM
   download code, as well as https://github.com/tock/tock/pull/2038, which fixes
   a HiFive1 kernel bug.
2. Disable the callback test. HiFive1 doesn't have a working timer driver. The
   callback test only worked before because of some other callback, which
   doesn't seem to occur with the new kernel.

Fixes #225
PR #226 (2020-08-04): Fix the broken Travis CI:
1. Update the `tock` submodule to a version that includes
   https://github.com/tock/tock/pull/2024, which fixes the OpenTitan BootROM
   download code, as well as https://github.com/tock/tock/pull/2038, which fixes
   a HiFive1 kernel bug.
2. Disable the callback test. HiFive1 doesn't have a working timer driver. The
   callback test only worked before because of some other callback, which
   doesn't seem to occur with the new kernel.

Fixes #225