tock/libtock-rs #317: c'e

0ble, 2021-07-20:
Trying to run some of the examples on microbit using a layout file from before the stack_size change. Tried increasing the stack in the microbit main file, but that does not seem it. Potentially relevant: grant_size = alloc_size on App Status from the process_standard message. Still working with c examples (using openocd for both).

jrvanwhy, 2021-07-20:
Which kernel version and `libtock-rs` version are you using? I'm trying to rule out a Tock 1.0 vs. 2.0 issue.

0ble, 2021-07-20:
Thanks for the quick response -- still using 1.0 and libtock-rs (0.2.0)

jrvanwhy, 2021-07-20:
Tock release 1.6 (the last 1.x release) does not have Microbit support, as far as I can tell. What commit are you using for the Tock kernel?

0ble, 2021-07-20:
8b8c54aeec62322a9366e3f87a772775da5770a9 (1.6).

0ble, 2021-07-20:
In case unclear, working fine with libtock-c, and, worked fine with libtock-rs pre stack_size change.

jrvanwhy, 2021-07-20:
> 8b8c54aeec62322a9366e3f87a772775da5770a9 (1.6).

I don't see a commit with that ID: [https://github.com/tock/tock/commit/8b8c54aeec62322a9366e3f87a772775da5770a9](https://github.com/tock/tock/commit/8b8c54aeec62322a9366e3f87a772775da5770a9).

0ble, 2021-07-20:
Oops [0c8c7127a0e820966a5336c79d961846765bd926](https://github.com/tock/tock/commit/0c8c7127a0e820966a5336c79d961846765bd926) apologies

jrvanwhy, 2021-07-20:
> Oops [0c8c7127a0e820966a5336c79d961846765bd926](https://github.com/tock/tock/commit/0c8c7127a0e820966a5336c79d961846765bd926) apologies

That is a Tock 2.0 kernel. The version numbers in the repository have not been updated, which I realize may be misleading.

The `master` branch changed from Tock 1.0 to Tock 2.0 when https://github.com/tock/tock/pull/2446 was merged. You'll need to use a kernel version from before that. It appears the Microbit was added a couple months before that was merged, so you may be able to find a working version.

0ble, 2021-07-20:
oh okay yeah the logs from process_standard said 1.6 so that is why. so microbit is not supported for 2.0? Then why does it work with libtock-c same version?

jrvanwhy, 2021-07-20:
> so microbit is not supported for 2.0?

It looks like it will be supported in Tock 2.0, when Tock 2.0 is released.

I don't know much about `libtock-c`, as I have barely worked with it. I'll let someone else answer there.

0ble, 2021-07-20:
Okay, thanks! Out of curiosity are the non-FPGA boards that do already work with libtock2?

jrvanwhy, 2021-07-20:
> Okay, thanks! Out of curiosity are there non-FPGA boards that do already work with libtock2?

As far as I am aware, `libtock2` has only been tested on HiFive1 rev b in QEMU, not on any physical boards. However, it should theoretically work on any board. The issue is not brokenness, but limited functionality (the only system call that has been implemented is Command, and no drivers have been implemented).

0ble, 2021-07-20:
Assuming some basic allow/subscribe, do you have any suggestions for writing user land drivers? Thinking about porting some of the ones that are in libtock-c for example

jrvanwhy, 2021-07-20:
Do you mean for the Tock 2.0 `libtock-rs`? No, I'm still trying to get the basic framework in place. I'm planning to write a document on how to write a `libtock-rs` driver once everything is in place.

hudson-ayers, 2021-07-20:
`libtock-c` has already moved to target the 2.0-alpha interface, which is why the examples there work against the 2.0-alpha kernel commit that you were using. Apologies for the version confusion, hopefully that should be resolved soon once the official 2.0 release happens. It will still be a little while after that before libtock-rs is updated to be usable with Tock 2.0

0ble, 2021-07-21:
> Do you mean for the Tock 2.0 libtock-rs

Yes. 

> I'm planning to write a document on how to write a libtock-rs driver once everything is in place.

Awesome.

> Apologies for the version confusion, hopefully that should be resolved soon once the official 2.0 release happens

No worries. Thanks to both of you for responding quickly!

closed 2021-07-21 by 0ble