tock/libtock-rs #319: Add documentation generation and hosting

jrvanwhy, 2021-07-26:
I've been writing doc comments in `libtock-rs` for years, but I almost never look at them. There's not a convenient way to view the generated documentation during code review.

If doing so is not unreasonably difficult, we should have some sort of automation that runs on each pull request that builds the documentation and uploads it for review (preferably with a comment on the PR with a link to the generated doc).

jrvanwhy, 2021-07-26:
@hudson-ayers Is there prior art here in the Tock project? I suspect there is.

hudson-ayers, 2021-07-26:
There is prior art, netlify automatically generates docs on each merged PR and publishes them to docs.tockos.org . It should be possible to do the same here. @ppannuto is the most familiar with the infrastructure I believe.

ppannuto, 2021-07-28:
Yeah, there's not a whole lot of magic really. Netlify is your standard CI container environment that runs arbitrary commands. For the kernel specifically, it calls https://github.com/tock/tock/blob/master/tools/netlify-build.sh, which basically just installs the toolchain then does a `make alldoc` then published the documentation folder as docs.tockos.org.

[image]

We have a Tock "team" on Netlify. I'm happy to add you (DM me preferred email for Netlfiy) and we can add another site there. It currently builds the kernel docs, the tock book, and the main tock website.