tock/libtock-rs #362: `.text` section misplaced in TBF files

jrvanwhy, 2022-01-20:
The `libtock2` examples have a misplaced `.text` section, resulting in "illegal instruction" faults. I've found the issue and will send a PR soon, but I wanted to do a write-up on my debugging process for future use. Because my writeup depends on a lot of data that currently only exists in my shell history, I'm writing this up as an issue (and may send a PR to put it into `docs/` in the future).

## The failure

When I attempt to run the `low_level_debug` example, I get the following app fault (note: the early debug messages are from a work-in-progress PR, but should be self-explanatory):

```
Package name: "low_level_debug"
TAB path: target/riscv32imac-unknown-none-elf/release/examples/low_level_debug.tab
Protected region size: 72
Found .stack section, size: 256
ELF file: "target/riscv32imac-unknown-none-elf/release/examples/low_level_debug"
TBF path: target/riscv32imac-unknown-none-elf/release/examples/low_level_debug.tbf
elf2tab command: "elf2tab" "-n" "low_level_debug" "-o" "target/riscv32imac-unknown-none-elf/release/examples/low_level_debug.tab" "--protected-region-size" "72" "--stack" "256" "target/riscv32imac-unknown-none-elf/release/examples/low_level_debug" "-v"
Spawning elf2tab
Creating "target/riscv32imac-unknown-none-elf/release/examples/low_level_debug.tbf"
Min RAM size from segments in ELF: 0 bytes
Number of writeable flash regions: 0
Entry point is in .start section
  Adding .start section. Offset: 72 (0x48). Length: 118 (0x76) bytes.
  Adding .text section. Offset: 190 (0xbe). Length: 64 (0x40) bytes.
Warning! Placing section .text at 0xbe, which is not 4-byte aligned.
Searching for .rel.X sections to add.
TBF Header:
               version:        2        0x2
           header_size:       64       0x40
            total_size:      512      0x200
                 flags:        1        0x1

        init_fn_offset:       40       0x28
        protected_size:        8        0x8
      minimum_ram_size:     2304      0x900

     start_process_ram: 2147493888 0x80002800
   start_process_flash: 537133128 0x20040048

elf2tab finished. exit status: 0
QEMU command: "tock2/tools/qemu/build/qemu-system-riscv32" "-M" "sifive_e,revb=true" "-kernel" "tock2/target/riscv32imac-unknown-none-elf/release/hifive1" "-nographic" "-device" "loader,file=target/riscv32imac-unknown-none-elf/release/examples/low_level_debug.tbf,addr=0x20040000"
Spawning QEMU
HiFive1 initialization complete.
Entering main loop.

panicked at 'Process low_level_debug had a fault', kernel/src/process.rs:1037:17
        Kernel version release-1.6-954-g17e698e8f

---| No debug queue found. You can set it with the DebugQueue component.

---| RISC-V Machine State |---
Last cause (mcause): Illegal instruction (interrupt=0, exception code=0x00000002)
Last value (mtval):  0x00000000

System register dump:
 mepc:    0x20012948    mstatus:     0x00000088
 mcycle:  0xFF33A41C    minstret:    0xFF33CBEA
 mtvec:   0x20010100
 mstatus: 0x00000088
  uie:    false  upie:   false
  sie:    false  spie:   false
  mie:    true   mpie:   true
  spp:    false
 mie:   0x00000888   mip:   0x00000000
  usoft:  false               false 
  ssoft:  false               false 
  msoft:  true                false 
  utimer: false               false 
  stimer: false               false 
  mtimer: true                false 
  uext:   false               false 
  sext:   false               false 
  mext:   true                false 
[... 295 lines cut ...]
.text (misplaced!):
200400be  09 47
200400c0  21 45 05 46 89 45 81 46  73 00 00 00 09 47 8d 46
200400d0  21 45 09 46 8d 45 73 00  00 00 82 80 41 11 06 c6
200400e0  97 00 00 00 e7 80 e0 fd  19 47 01 45 81 45 73 00
200400f0  00 00 00 00 97 00 00 00  e7 80 80 fe 00 00 00 00
20040100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040180  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
20040190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
200401f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
```

`.text` in the TBF starts at `0x200400be`, but in the ELF file it starts at `0x200400c0`! There should be 2 more bytes of padding between `.start` and `.text`, so that `.text` is located at a multiple of 4 bytes.

jrvanwhy, 2022-01-20:
@hudson-ayers @alistair23 @alexandruradovici 

Do you think it would be useful for me to clean up the initial post and add it to the `docs/` directory, or is the debugging process too specific to this particular issue to be helpful?

hudson-ayers, 2022-01-20:
I think this would be helpful to anyone debugging layout / entry point bugs! But I wouldn't put too much effort into cleaning it up, its pretty readable as-is, you can add a disclaimer that describes what it was originally written for

dcz-self, 2022-03-07:
It's already proved helpful for someone debugging entry point bugs. I'm looking forward to including it in docs.

commit 18e04385 (2022-06-28): Add a process fault debugging example to `doc/`.

Closes #362
closed 2022-06-28 by bors[bot]