tock/libtock-rs #400: Investigate new Miri strict provenance flags

jrvanwhy, 2022-04-02:
Miri has a new strict provenance flag, which we may want to enable: https://github.com/rust-lang/miri/pull/2045

Open question: Is `-Zmiri-strict-provenance` strictly more restrictive than Miri's defaults (Stacked Borrows without `-Zmiri-tag-raw-pointers`)? If yes, perhaps we can deduplicate the Miri invocations to make `make test` faster?

Gankra, 2022-04-03:
Yes it's strictly stricter, and implies -Zmiri-tag-raw-pointers too.

jrvanwhy, 2022-04-04:
> Yes it's strictly stricter, and implies -Zmiri-tag-raw-pointers too.

Fantastic, thank you for confirming!

jrvanwhy, 2022-06-11:
`-Zmiri-strict-provenance` works fine for our unit tests. I'll send a PR once the [toolchain update](https://github.com/tock/libtock-rs/pull/409) is merged.

@Gankra I have some questions about how strict provenance interacts with FFI. Where is the most appropriate place to ask them?

commit 277a7f68 (2022-06-14): Merge #410

410: Enable strict provenance in Miri. r=jrvanwhy a=jrvanwhy

This merges the two miri commands together, as strict provenance is strictly more restrictive than all other configurations.

See https://github.com/rust-lang/miri/pull/2045 for some more background on `-Zmiri-strict-provenance`.

Closes #400
PR #410 (2022-06-14): Enable strict provenance in Miri.
This merges the two miri commands together, as strict provenance is strictly more restrictive than all other configurations.

See https://github.com/rust-lang/miri/pull/2045 for some more background on `-Zmiri-strict-provenance`.

Closes #400
closed 2022-06-14 by bors[bot]