tock/libtock-rs #422: `blink` example doesn't exist

lqwk, 2022-07-20:
The line `make flash-hail EXAMPLE=blink # Flash the example 'blink' program to the hail platform` doesn't work because the the [examples](https://github.com/tock/libtock-rs/tree/master/examples) folder doesn't contain `blink`

```
$ make flash-microbit_v2 EXAMPLE=blink
LIBTOCK_PLATFORM=microbit_v2 cargo run --example blink  \
                --target=thumbv7em-none-eabi --release -- --deploy=tockloader
error: no example target named `blink`
make: *** [flash-microbit_v2] Error 101
```

`$ make flash-microbit_v2 EXAMPLE=leds` works though

jrvanwhy, 2022-07-21:
`README.md` still needs to be updated after `libtock-rs` was substantially rewritten.

hudson-ayers, 2022-07-21:
Closing as a duplicate of https://github.com/tock/libtock-rs/issues/398

closed 2022-07-21 by hudson-ayers