tock/libtock-rs #431: Cannot compile examples for Raspberry Pi Pico

m1cha1s, 2022-09-04:
When I try to compile the any example with the commands specified in README it fails with [this](https://github.com/tock/libtock-rs/files/9485230/dump.txt).

alexandruradovici, 2022-09-04:
We do not yet support libtock-rs for the Raspberry Pi Pico. From what I can read in the error, it seems that you are missing the armv6 toolchain.

Please try adding it using `rustup target add thumbv6m-none-eabi` before running cargo.

jrvanwhy, 2022-09-06:
Looks to me like we do support the Raspberry Pi Pico: https://github.com/tock/libtock-rs/blob/master/runtime/layouts/raspberrypi_pico.ld

alexandruradovici, 2022-09-06:
It used to build when I added it, but I think I had the `armv6` toolchain already installed.

tommy4111, 2022-10-08:
Got this failures after doing `rustup target add thumbv6m-none-eabi `  
[dump.txt](https://github.com/tock/libtock-rs/files/9740170/dump.txt).  Any ideas?

error: could not compile `syn` due to previous error__