tock/libtock-rs #479: Cannot get Microbit_v2 examples to work

willAlbertini707, 2023-07-26:
I can't seem to get a response from the microbit v2 board when running the console example (or get any example to work for that matter). There is no problem making the app and establishing a connection with the board. It compiles fine but there is no "Hello World" message or led flashing to verify that data is being sent. Does anyone have a suggestion? Thanks!

-->make flash-microbit_v2 EXAMPLE=console 
LIBTOCK_PLATFORM=microbit_v2 cargo run --example console  \
		--target=thumbv7em-none-eabi --release -- --deploy=tockloader
    Finished release [optimized + debuginfo] target(s) in 0.03s
     Running `cargo run -p runner --release target/thumbv7em-none-eabi/release/examples/console --deploy=tockloader`
    Finished release [optimized + debuginfo] target(s) in 0.04s
     Running `target/release/runner target/thumbv7em-none-eabi/release/examples/console --deploy=tockloader`
[INFO   ] Using openocd channel to communicate with the board.
[INFO   ] Using settings from KNOWN_BOARDS["microbit_v2"]
[STATUS ] Preparing to uninstall apps...
[WARNING] Unknown TLV block in TBF header.
[WARNING] You might want to update tockloader.
[INFO   ] Only one app on board.
[STATUS ] Attempting to uninstall:
[STATUS ]   - console
[STATUS ] Uninstall complete.
[INFO   ] Finished in 1.587 seconds
[INFO   ] Using openocd channel to communicate with the board.
[INFO   ] Using settings from KNOWN_BOARDS["microbit_v2"]
[STATUS ] Installing app on the board...
[WARNING] Unknown TLV block in TBF header.
[WARNING] You might want to update tockloader.
[INFO   ] Found sort order:
[INFO   ]   App "console" at address 0x40060
[INFO   ] Installing app bundle. Size: 8192 bytes.
[INFO   ] Finished in 2.618 seconds
[WARNING] Unknown arguments passed. You may need to update tockloader.
[WARNING] Unknown argument "--bundle-apps"
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 4 serial ports.
Multiple serial port options found. Which would you like to use?
[0]	/dev/cu.URT0 - n/a
[1]	/dev/cu.BeatsStudio - n/a
[2]	/dev/cu.BeatsStudio2 - n/a
[3]	/dev/cu.usbmodem14202 - BBC micro:bit CMSIS-DAP - mbed Serial Port

Which option? [0] 3
[INFO   ] Using "/dev/cu.usbmodem14202 - BBC micro:bit CMSIS-DAP - mbed Serial Port".
[INFO   ] Listening for serial output.
Initialization complete. Entering main loop.
tock$ status
Total processes: 0
Active processes: 0
Timeslice expirations: 0
tock$

bradjc, 2023-08-02:
This looks like the kernel couldn't allocate a RAM region for the app.

I had to modify the microbit linker script:

```diff
--- a/runtime/layouts/microbit_v2.ld
+++ b/runtime/layouts/microbit_v2.ld
@@ -5,7 +5,7 @@ TBF_HEADER_SIZE = 0x60;
 FLASH_START = 0x00040000;
 FLASH_LENGTH = 256K;

-RAM_START = 0x20004000;
+RAM_START = 0x20006000;
 RAM_LENGTH = 112K;

 INCLUDE libtock_layout.ld
```

But I was able to get the console app to run on microbit.

bradjc, 2023-08-22:
This should be fixed with `make tab`.

closed 2023-08-22 by bradjc