tock/libtock-rs #492: QEMU Setup Issue

brghena, 2023-08-04:
Something seems to be wrong with the QEMU setup. When I run `make setup` in a fresh libtock-rs installation, it fails the very first step (`setup-qemu`) the first time I run it. The second time I run it though, it says everything is fine and continues on to the other steps. Afterwards though, running `make qemu-example` also fails.

I suspect that I'm missing some dependency for QEMU, but the important parts here are:
1. Documenting what that dependency is
2. Figuring out why it behaves differently when run twice in a row
3. Determining if `make setup` should block on `setup-qemu` succeeding

Since the error is about `pixman`, I tried `sudo apt install libpixman-1.0`, but that didn't change the situation.

Here's the log:

### First time
```
[brghena@ubuntu libtock-rs] [master] $ make setup
CI=true make -C tock ci-setup-qemu
make[1]: Entering directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock'
cksum: tools/qemu-runner/opentitan-boot-rom.elf: No such file or directory

********************************************************************************
* CI-Setup: Build QEMU                                                         *
********************************************************************************

Building QEMU, this could take a few minutes

Note: switching to '0ebf76aae58324b8f7bf6af798696687f5f4c2a9'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 0ebf76aae5 Merge tag 'nvme-next-pull-request' of git://git.infradead.org/qemu-nvme into staging
Using './build' as the directory for build output
Submodule 'dtc' (https://gitlab.com/qemu-project/dtc.git) registered for path 'dtc'
Submodule 'meson' (https://gitlab.com/qemu-project/meson.git) registered for path 'meson'
Submodule 'slirp' (https://gitlab.com/qemu-project/libslirp.git) registered for path 'slirp'
Submodule 'tests/fp/berkeley-softfloat-3' (https://gitlab.com/qemu-project/berkeley-softfloat-3.git) registered for path 'tests/fp/berkeley-softfloat-3'
Submodule 'tests/fp/berkeley-testfloat-3' (https://gitlab.com/qemu-project/berkeley-testfloat-3.git) registered for path 'tests/fp/berkeley-testfloat-3'
Submodule 'ui/keycodemapdb' (https://gitlab.com/qemu-project/keycodemapdb.git) registered for path 'ui/keycodemapdb'
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/dtc'...
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/meson'...
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/slirp'...
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/tests/fp/berkeley-softfloat-3'...
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/tests/fp/berkeley-testfloat-3'...
Cloning into '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/ui/keycodemapdb'...
The Meson build system
Version: 0.59.3
Source dir: /home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu
Build dir: /home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/build
Build type: native build
Project name: qemu
Project version: 7.0.50
C compiler for the host machine: cc -m64 -mcx16 (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc -m64 -mcx16 ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program scripts/symlink-install-tree.py found: YES (/usr/bin/python3 /home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/scripts/symlink-install-tree.py)
Program sh found: YES (/usr/bin/sh)
Program python3 found: YES (/usr/bin/python3)
WARNING: Broken python installation detected. Python files installed by Meson might not be found by python interpreter.
Program iasl found: NO
C++ compiler for the host machine: c++ -m64 -mcx16 (gcc 9.4.0 "c++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C++ linker for the host machine: c++ -m64 -mcx16 ld.bfd 2.34
Program cgcc found: NO
Library m found: YES
Run-time dependency threads found: YES
Library util found: YES
Run-time dependency appleframeworks found: NO (tried framework)
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency gio-2.0 found: YES 2.64.6
Program /usr/bin/gdbus-codegen found: YES (/usr/bin/gdbus-codegen)
Run-time dependency gio-unix-2.0 found: YES 2.64.6
Run-time dependency pixman-1 found: NO (tried pkgconfig)

../meson.build:522:2: ERROR: Dependency "pixman-1" not found, tried pkgconfig

A full log can be found at /home/brghena/Dropbox/repos/tock/libtock-rs/tock/tools/qemu/build/meson-logs/meson-log.txt

ERROR: meson setup failed

make[1]: *** [Makefile:547: ci-setup-qemu] Error 1
make[1]: Leaving directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock'
make: *** [Makefile:53: setup-qemu] Error 2
```

### Second Time
```
[brghena@ubuntu libtock-rs] [master] $ make setup
CI=true make -C tock ci-setup-qemu
make[1]: Entering directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock'
bash: line 3: ./config.status: No such file or directory
make[1]: *** No rule to make target 'config-host.mak', needed by 'meson.stamp'.  Stop.
cksum: tools/qemu-runner/opentitan-boot-rom.elf: No such file or directory

********************************************************************************
* CI-Setup: Get OpenTitan boot ROM image                                       *
********************************************************************************

# Download OpenTitan image. The latest image URL is available at
# https://storage.googleapis.com/artifacts.opentitan.org/latest.txt
# We download a fixed version so that new OpenTitan images do not
# unexpectedly change OpenTitan's behavior in our CI.
Downloading OpenTitan boot ROM
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 10.2M  100 10.2M    0     0  19.0M      0 --:--:-- --:--:-- --:--:-- 19.0M
make[1]: Leaving directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock'
rustup install stable
info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'

  stable-x86_64-unknown-linux-gnu unchanged - rustc 1.71.1 (eb26296b5 2023-08-03)

info: checking for self-update
cargo +stable install elf2tab
    Updating crates.io index
     Ignored package `elf2tab v0.11.0` is already installed, use --force to override
cargo miri setup
A libstd for Miri is now available in `/home/brghena/.cache/miri/HOST`.
rustup target add --toolchain stable thumbv7em-none-eabi
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
```

### Running qemu-example
```
[brghena@ubuntu libtock-rs] [master] $ make qemu-example EXAMPLE=console
make -C tock/boards/hifive1 \
	/home/brghena/Dropbox/repos/tock/libtock-rs/tock/target/riscv32imac-unknown-none-elf/release/hifive1.elf
make[1]: Entering directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/boards/hifive1'
    Finished release [optimized + debuginfo] target(s) in 0.15s
   text	   data	    bss	    dec	    hex	filename
  66656	     12	   6128	  72796	  11c5c	/home/brghena/Dropbox/repos/tock/libtock-rs/tock/target/riscv32imac-unknown-none-elf/release/hifive1
make[1]: Leaving directory '/home/brghena/Dropbox/repos/tock/libtock-rs/tock/boards/hifive1'
LIBTOCK_PLATFORM="hifive1" cargo run --example "console" -p libtock \
	--release --target=riscv32imac-unknown-none-elf -- --deploy qemu
    Finished release [optimized + debuginfo] target(s) in 0.06s
     Running `cargo run -p runner --release target/riscv32imac-unknown-none-elf/release/examples/console --deploy qemu`
    Finished release [optimized + debuginfo] target(s) in 0.06s
     Running `target/release/runner target/riscv32imac-unknown-none-elf/release/examples/console --deploy qemu`
Overriding command-line specified protected_region_size with tbf_protected_region_size symbol = 96 bytes!
thread 'main' panicked at 'failed to spawn QEMU: Os { code: 2, kind: NotFound, message: "No such file or directory" }', runner/src/qemu.rs:35:18
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted (core dumped)
make: *** [Makefile:77: qemu-example] Error 134
```

alistair23, 2023-08-04:
> `sudo apt install libpixman-1.0`

Debian split its dev packages, so you probably need `libpixman-1.0-dev` (or something like that)

alistair23, 2023-08-04:
> * Figuring out why it behaves differently when run twice in a row

The script for some reason is detecting that you already have QEMU built (even though it failed) so it's continuing on. I guess the check there needs to be updated to catch this case and report an error

brghena, 2023-08-04:
I confirm that worked. So `libpixman-1.0-dev` is a requirement of the QEMU stuff, and therefore a requirement of libtock-rs as currently designed.

bradjc, 2023-08-08:
Why not split setup qemu into its own target? Right now if you don't want to set all of this up you can't use `make setup` as the instructions indicate.