tock/libtock-rs #502: Glob copying issue

stevefan1999-personal, 2023-08-14:
It seems like
https://github.com/tock/libtock-rs/blob/ae66d0ae0a6e979f42b7ab12226bc83ba8aa48c8/Makefile#L142-L143
I guess this may work on zsh but it is not supported on bash.

Easy to mitigate, we can just split the copy into two different command.

ppannuto, 2023-08-14:
No big deal to split, but I run (stock) bash, and it seems to work fine:
```
$ touch a.b a.c; mkdir d; cp a.{b,c} d/; tree; echo $(BASH_VERSION)
.
├── a.b
├── a.c
└── d
    ├── a.b
    └── a.c

2 directories, 4 files
5.2.15(1)-release
```

What version doesn't this work on?

jrvanwhy, 2023-08-14:
> Another option is to define which shell to use in the Makefile by adding `SHELL=/bin/bash`.

@stevefan1999-personal would you mind testing whether this fixes your issue? My system uses Bash by default and I don't want to change that. If it fixes your issue, please send a PR with the fix.

stevefan1999-personal, 2023-08-15:
> > Another option is to define which shell to use in the Makefile by adding `SHELL=/bin/bash`.
> 
> @stevefan1999-personal would you mind testing whether this fixes your issue? My system uses Bash by default and I don't want to change that. If it fixes your issue, please send a PR with the fix.

Yes, it seems like I was using dash all the time despite `echo $0` said I'm using bash

commit e542dd22 (2023-08-21): Specify Bash as the shell to use in Makefile.

Closes #502.
PR #504 (2023-08-21): Specify Bash as the shell to use in Makefile.
Closes #502.
closed 2023-08-21 by bradjc