tock/libtock-rs #521: tockloader: failing to load libtock-rs apps

twilfredo, 2023-11-06:
# Issue

When loading the example apps (blink, console etc...) from libtock-rs with tockloader. I see the following error. The particular error below is for the gpio example from libtock-rs loading into the `nano33ble`, so built with `nrf52840` platform with `make nrf52840 EXAMPLE=gpio`. Then `tockloader install`. The nanoble33 was in bootloader mode prior to running the below command, I tried without this aswell...same issue.

```
(rust:1.70) ↪ tockloader install
[INFO   ] No TABs passed to tockloader.
[STATUS ] Searching for TABs in subdirectories.
 Which TAB files do you want to use? [./target/tbf/nrf52840/gpio.
tab]
[INFO   ] Using: ['./target/tbf/nrf52840/gpio.tab']
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/ttyACM0 - Nano 33 BLE - TockOS".
[INFO   ] Using serial channel to communicate with the board.
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] Using "/dev/ttyACM0 - Nano 33 BLE - TockOS".
[STATUS ] Installing app on the board...
[INFO   ] Waiting for the bootloader to start
[ERROR  ] Could not meet fixed address requirements.
[ERROR  ] Unable to find a valid sort order to flash apps.
```

I have tried on 3 different nRF boards, nrf52DK/Particle Boron/Nano33BLE.  What's likely the culprit here?

Note that libtock-c apps seems to work as intended, I have been manually loading them with `tockloader install *.tab` for the `nano33ble`, and the app being loaded with the process console/tockloader.

jrvanwhy, 2023-11-06:
There is a mismatch between the address ranges the app was compiled for and the address ranges supported by the kernel you are running. The reason this works with libtock-c is because libtock-c apps support PIC (note that all 3 boards you listed are ARM-based), and can therefore be placed at any address.

The address ranges that `libtock-rs` uses by default are defined in [`libtock_build_scripts::PLATFORMS`](https://github.com/tock/libtock-rs/blob/21b3c05dc4ad602a9e158cb4ab9e3bfaac2c29c1/build_scripts/src/lib.rs#L5). Check the address range for the value of `LIBTOCK_PLATFORM` you used against the kernel image you're using; I suspect a mismatch there.

Unfortunately the address ranges for apps can change with kernel version (as the kernel itself grows and shrinks), so you may end up needing to manually pass the address ranges via environment variables. Check the comments in the file I linked above for instructions.

twilfredo, 2023-11-07:
Thanks! that was it, sent a PR that fixes this for the nrf52840 based boards. 

I can get a single app to work. But not more than one at a time, even if they are manually compiled for the next available flash address and loaded (with tockloader) they will not execute.  See below, only blink is running.

```
[INFO   ] Using settings from KNOWN_BOARDS["nrf52dk"]
[STATUS ] Installing app on the board...
[INFO   ] Found sort order:
[INFO   ]   App "blink" at Flash=0x40080
[INFO   ]   App "console" at Flash=0x42080
[INFO   ] App Layout:
[INFO   ]    0x40000┬──────────────────────────────────────────────────┐
[INFO   ]           │App: blink                             [Installed]│
[INFO   ]           │  Length: 8192 (0x2000)                           │
[INFO   ]    0x42000┼──────────────────────────────────────────────────┤
[INFO   ]           │App: console                            [From TAB]│
[INFO   ]           │  Length: 8192 (0x2000)                           │
[INFO   ]    0x44000┴──────────────────────────────────────────────────┘
[INFO   ] Finished in 3.787 seconds
```

I did see https://github.com/tock/libtock-rs/issues/28 is there any progress on this? What still needs to be done?

twilfredo, 2023-11-07:
Woops, Turns out I had to change the RAM address too to, suppose there was an overlap but silently failed? I have multiple apps working now.

```
[INFO   ] Using settings from KNOWN_BOARDS["nrf52dk"]
[STATUS ] Installing app on the board...
[INFO   ] Found sort order:
[INFO   ]   App "blink" at Flash=0x40080
[INFO   ]   App "console" at Flash=0x42080
[INFO   ]   App "buttons" at Flash=0x44080
[INFO   ] App Layout:
[INFO   ]    0x40000┬──────────────────────────────────────────────────┐
[INFO   ]           │App: blink                             [Installed]│
[INFO   ]           │  Length: 8192 (0x2000)                           │
[INFO   ]    0x42000┼──────────────────────────────────────────────────┤
[INFO   ]           │App: console                           [Installed]│
[INFO   ]           │  Length: 8192 (0x2000)                           │
[INFO   ]    0x44000┼──────────────────────────────────────────────────┤
[INFO   ]           │App: buttons                            [From TAB]│
[INFO   ]           │  Length: 16384 (0x4000)                          │
[INFO   ]    0x48000┴──────────────────────────────────────────────────┘
[INFO   ] Finished in 5.260 seconds
```

Serial

```
Particle Boron: Initialization complete. Entering main loop
button count: Hello world!
                          1
                           Particle Boron: Initialization complete. Entering main loop
button count: Hello world!
                          1
                           tock$ button 0: Pressed
                                                  button 0: Released
                                                                    button 0: Pressed
                                                                                     button 0: Released
```

closed 2023-11-10 by jrvanwhy