tock/libtock-rs #566: [ERROR  ] Mixing fixed address and position-independent apps is currently unsupported.

rabarar, 2025-01-03:
attempting to flash an example tab

successfully build an example:
```
 % make tab EXAMPLE=leds            
```

However, attempting to flash:
```
% tockloader install target/tab/leds.tab
```

results in:
```
[ERROR  ] Mixing fixed address and position-independent apps is currently unsupported.
```

lschuermann, 2025-01-03:
Can you try to run a `tockloader erase-apps` on your board, of post the output of `tockloader list`? It might be the case that you have a libtock-c app already installed on your board, which would be a position-independent app. libtock-rs (well, rustc) currently does not support the relocation mode that Tock would need for relocatable apps.

In general, it'd be great if you could post more details about your environment (which board are you using, etc.).

lschuermann, 2025-01-03:
Great to see it works! (I guess it must be a used(?) board if it already had a blink app pre-installed)

Unfortunately, you'll run into this error right now when trying to mix libtock-rs (fixed address) and libtock-c apps (relocatable) on Cortex-M platforms right now. This is nothing fundamental, but just not something we handle right now in Tockloader. I'll go ahead and close this issue, but feel free to reopen if anything's still unclear. Fixes to Tockloader are of course also always welcome. :smile:

closed 2025-01-03 by lschuermann

rabarar, 2025-01-03:
I loaded a tab from the web that must have been a c-generated tab (following the book)