tock/libtock-rs #577: nostack option for ARM system calls

Ioan-Cristian, 2025-07-27:
Should ARM system calls be marked `nostack` as [here](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_arm.rs#L58). The Rust reference [mentions](https://doc.rust-lang.org/reference/inline-assembly.html#r-asm.options.supported-options.nostack) that `nostack` must not be used if data is pushed on the stack. However, on ARMv*-M architectures, the hardware automatically pushes data on stack. Furthermore, the Tock kernel [does](https://github.com/tock/tock/blob/92ac4ace041ad21b1b48d6f183d06924e85da5fe/arch/cortex-m0/src/lib.rs#L219) use the process stack for storing process data on exception entry by setting the SPSEL bit to 1. I doubt the reference excludes hardware-related stack operations. As a consequence, the current `libtock-rs` results in undefined behaviour that just happens to work for the current LLVM backend. Note that  [libtock-c](https://github.com/tock/libtock-c/blob/24c0f3f5ff5f78cec936e9af9cf81a5fb0431b55/libtock/tock.c#L243) uses the `memory` clobber which covers all kinds of memory accesses, including the stack.

Am I missing something?

Ioan-Cristian, 2025-07-31:
@ppannuto I had a typo in my original post. I am sorry for it! The first sentence is a question:

Should ARM system calls be marked `nostack` as [here](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_arm.rs#L58)?

It should be interpreted as: why is the `nostack` option used for ARM system calls since the hardware touches the stack space below the stack pointer?

> Unless the nostack option is set, assembly code is allowed to use stack space below the stack pointer.

I understand this sentence as following:

* `nostack` is used => the software/the hardware is not allowed to use any stack space below the stack pointer.
* `nostack` is not used => the software/the hardware is allowed to use the stack space below the stack pointer, as long as it respects the requirements you cited.

Since all ARM system calls are marked `nostack`, they shouldn't touch any stack space below the stack pointer. However, as previously mentioned, the ARMv*-M architectures automatically push and pop data on the stack on exception entry, including system call exceptions. As a consequence, in my opinion, the current ARM system call implementation results in undefined behaviour that just happens to work.

ppannuto, 2025-08-04:
> the software/the hardware

I don't think it's appropriate to lump these together like this (on ARM at least). The _assembly code_ does not "use stack space below the stack pointer" — i.e., no assembly instruction explicitly touches the stack in any way.

Yes, the `svc` happens to cause _hardware_ to use stack space, but hardware also (1) validates, fixes, and restores stack pointer alignment if-needed, and (2) more generally restores the stack to an 'unchanged' state when it is done.

However, this is conceptually no different than if _any other hardware interrupt_, e.g. a timer interrupt, happened to fire while inside the assembly block. My understanding is that the `nostack` directive _only_ applies to software, because it would be meaningless / never-usable (without disabling interrupts every time you entered an assembly block) to make statements about what hardware does with the stack.

Similarly, the Tock kernel does assume that the userspace stack is in a sane, aligned state with contents in a known layout relative to the stack pointer — but that is safe/okay because the ARM _hardware_ ensures that is the case on exception entry.

Ioan-Cristian, 2025-08-04:
@ppannuto _My_ understanding of `nostack` is different.

First of all, `svc` does touch the area under the stack pointer. I see no difference between `svc` and x86 `call` instructions. In both cases, data is pushed on the stack. In both cases, the stack is touched as a consequence of the program's instruction flow.

It doesn't matter that `svc` pushes data, then pops it back. The [example](https://doc.rust-lang.org/reference/inline-assembly.html#r-asm.options.supported-options.nostack) provided by the Rust reference pushes, then pops data, but it is still marked as undefined behaviour.

The question that might arrise: why would the compiler use the stack area below the stack pointer? One situation I have managed to reproduce with `gcc` on a `x64` machine long time ago is the following: you have a large leaf function that is called in several places. You compile the code with `-fomit-frame-pointer` or `-fomit-leaf-frame-pointer`. When optimizations are enabled, the compiler ignores the frame pointer, but also the instructions for allocating and deallocating the stack: substraction of the SP in the prologue + addition of the SP in the epilogue. Why would the compiler skip stack allocation + deallocation? For several reasons:

1. The stack pointer may not need to be updated to use the space under the stack pointer. Instead of using positive offsets for stack objets, the compile can use negative offsets.
2. Omitting stack substraction + addition saves two instructions: less code + faster execution, especially on small architectures such as ARMv*-M architectures.

I have tried to reproduce this behaviour with `libtock-rs` by marking `syscall4()` `#[inline(never)]` and using `RUSTFLAGS="-C force-frame-pointers=no"`. Unfortunately, the compiler still creates a frame pointer as the `force-frame-pointers=no` does not guarantee that a frame pointer will not be generated. The result is the same whether the optimization level is set to `z` or `3`.

I'll try to think of an example where `svc` marked as `nostack` leads to erroneous code generation.

Curious what @jrvanwhy thinks about this.

jrvanwhy, 2025-08-06:
> I think there are separate concerns around buffers which have been `allow`ed to the kernel, and in which cases the Tock kernel can update those. At minimum, my instinct is that it is probably necessary to remove `nomem` from the Yield family of syscalls here — but perhaps that synchronization is already handled at another level in libtock-rs.

`nomem` is not currently set for Yield ([yield1 interface](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/platform/src/raw_syscalls.rs#L66), [yield2 interface](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/platform/src/raw_syscalls.rs#L87), [ARM yield1 impl](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_arm.rs#L5), [ARM yield2 impl](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_arm.rs#L25), [RISC-V yield1 impl](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_riscv.rs#L8), [RISC-V yield2 impl](https://github.com/tock/libtock-rs/blob/5b4d692956fe834ec734e2541d3354356a4d5ae0/runtime/src/syscalls_impl_riscv.rs#L41)).

> Yes, the `svc` happens to cause *hardware* to use stack space, but hardware also (1) validates, fixes, and restores stack pointer alignment if-needed, and (2) more generally restores the stack to an 'unchanged' state when it is done.

My mental model was that when the interrupt is complete, the register values and stack pointer value would be restored but the contents of the bytes below the stack pointer (those that the hardware used to save the register values) would not be restored. When the hardware pushes the register values to the stack, it is overwriting *something*, and I don't see how it can save the value of those bytes.

So it's not entirely 'unchanged', it's just that the compiler must guarantee that anytime an interrupt can happen, the stack is in a state where that modification is acceptable (i.e. those bytes are ignored).

You know how ARM works a lot better than me, though. Is my mental model correct or incorrect?

Also, I believe you are saying that the software interrupt mechanism (triggered by `svc`) does not change anything in userspace that the hardware interrupt mechanism does not, correct?

> My understanding is that the `nostack` directive *only* applies to software, because it would be meaningless / never-usable (without disabling interrupts every time you entered an assembly block) to make statements about what hardware does with the stack.

The language reference can't really distinguish hardware from the assembly code. I was also under this understanding (and see some support for it [in related discussions](https://rust-lang.zulipchat.com/#narrow/channel/216763-project-inline-asm/topic/Are.20rustix.20syscall.20functions.20sound.20.3F). However, I would hope to see some statement that the compiler will still leave the stack in a configuration that allows for interrupts, but I don't see that.

I *think* we're fine here, but I'm not 100% sure of that. If my mental model and understanding correct, then I can go upstream and ask whether `nostack` makes enough guarantees to allow for interrupts to execute without causing UB or faulting the process.

ppannuto, 2025-08-07:
> The [example](https://doc.rust-lang.org/reference/inline-assembly.html#r-asm.options.supported-options.nostack) provided by the Rust reference pushes, then pops data, similar to the behaviour of svc, but it is still marked as undefined behaviour.

The example linked here uses Intel's `push` assembly instruction, which according to the [Intel reference manual; PDF page 1256](https://cdrdv2-public.intel.com/835757/325383-sdm-vol-2abcd.pdf) is a _software instruction that explicitly modifies the stack_.

Contrast this to ARM's `svc` instruction, which according to the [ARM reference manual](https://developer.arm.com/documentation/dui0801/f/A32-and-T32-Instructions/SVC), "Operation: The `SVC` instruction causes an exception.", i.e., the `SVC` instruction makes no direct assumptions about the state or behavior of the stack.

---

> So it's not entirely 'unchanged', it's just that the compiler must guarantee that anytime an interrupt can happen, the stack is in a state where that modification is acceptable (i.e. those bytes are ignored).

Yes, that's exactly the key point here.

**On ARM platforms**, it is either the case that interrupts must be disabled before before entering _any_ assembly block with the `nostack` directive set OR it is safe to use `svc` inside of an assembly block with the `nostack` directive set (because **on ARM**, `svc` behaves just like any other interrupt—e.g. a timer interrupt—that happens to fire during execution of the assembly).

I am not an expert on Intel or RISC-V platforms, and cannot speak to how `nostack` should (or should not) be used on those platforms.

ppannuto, 2025-08-08:
From the chat the Johnathan linked, I think [this comment](https://rust-lang.zulipchat.com/#narrow/channel/216763-project-inline-asm/topic/Are.20rustix.20syscall.20functions.20sound.20.3F/near/271887270) is particularly relevant:

> I think these are still not considered "use of stack" though, because redzones are respected. They are not very different from signal handlers' use of stack.

IIUC, `redzone`s are the general compiler concept for "space that may be clobbered by interrupt handlers", and this chat suggests that Rust's `nostack` still respects `redzone` rules — which is to say that I suspect that `nostack` is safe to use on all platforms.

jrvanwhy, 2025-08-11:
I asked about this at https://users.rust-lang.org/t/asm-s-nostack-and-arm-interrupts/132881, hopefully we get a clearer answer there.