tock/libtock-rs #578: Implementation of streaming process slice

Ioan-Cristian, 2025-07-27:
The new Ethernet TAP driver uses streaming process slices. I have a minimal implementation [here](https://github.com/OxidosAutomotive/libtock-rs/blob/streaming_process_slice/platform/src/streaming_process_slice.rs). However, I was unable to use the standard `libtock-rs` system calls (TockSyscalls), so raw system calls are used instead. My use case is `libtock-rs` + `smoltcp`.

`Handle<AllowRw<...>>` forces the code to allow than disallow the buffer when the handle goes out of scope. Furthermore, depending on the configuration, it even checks whether the pointer and the length returned by the disallow are null, which is not true in the case of the streaming process slice, which constantly swaps between two buffers. I have tried a theoretical implementation of the streaming process slice with two `Handle<Allow<...>>`, but the code became quickly too contrived. I decided to give up, since that implementation is anyway inefficient.

Is my understanding of the `libtock-rs` too poor? Can streaming process slices be implemented efficiently using TockSyscalls, and thus avoid unsafe code?

alexandruradovici, 2025-07-27:
Can you detail the issue that is preventing you from using the system calls?

jrvanwhy, 2025-08-11:
Our Allow APIs definitely need "swap" functionality in addition to their "allow" and "unallow" functions.

It's not immediately clear to me whether adding swap to the existing `share` API would solve your problem, or if this would require a Pin-based Allow API.